balena-io/etcher #3312: UI unusable by weird Scrambling

Defencelord2222, 2020-10-07:
Etcher version: 1.5.109
Operating system and architecture: MacOS 10.15.07
Image flashed: N/A
Do you see any meaningful error information in the DevTools? N/A

Straight After installing The application the UI is unusable. Please see below below link for image:

https://gyazo.com/965fd3dce49abda6445b696ec4cb4fb9

thundron, 2020-10-13:
I have no idea about that, could you try an older version perhaps? I wouldn't know how far back though

shawaj, 2020-12-13:
Copying image here for reference:
[image]

Can you try with the latest version v1.5.112 @Defencelord2222 ?