balena-io/etcher #3314: Loosing data while copy from one SD card to another

adasiko, 2020-10-08:
- **Etcher version: v1.5.109** 
- **Operating system and architecture: Windows 10 64 bit** 
- **Image flashed: Ubuntu Server for Raspberry Pi** 
- **Do you see any meaningful error information in the DevTools?: no**

Source SD card is 16 GB and destination SD card is 32 GB.
Cloning process and verification successfully anded and source SD card partition is empty and destination SD is too...

JoschaHemlep, 2020-11-14:
I have the same problem, with the same Etcher version, same Operating system. On the source SD Card 16gb was/is raspbian (buster), the target 128gb card was empty. Source card doesn't boot anymore :-( 
No errors/warning during cloning and validation process.

Is there a chance to recover the source card without formatting?

zvin, 2020-11-23:
@adasiko @JoschaHemlep there was a bug in Etcher 1.5.106 - 1.5.110 that remove the partition table on Windows.
If you know how your SD card was partitioned, you can recover the data by recreating the partition table.

lurch, 2020-11-24:
> there was a bug in Etcher 1.5.106 - 1.5.110 that removed the partition table on Windows.

Ooops! :cry:  I guess that explains a few of the "ranty" bug-reports that have been seen.

adasiko, 2020-12-02:
Version 1.5.111: works as expected.

closed 2020-12-02 by adasiko