balena-io/etcher #3316: a bootable USB stick

zhang2514yuchi, 2020-10-10:
- **Etcher version:** 
- **Do you see any meaningful error information in the DevTools?** 
[https://cdn.luogu.com.cn/upload/image_hosting/7m3qgj6e.png](url)
I installed the boot USB disk according to the tutorial in the official document, but there was an error as shown in the figure

zhang2514yuchi, 2020-10-10:
[
[image]
](url)

zhang2514yuchi, 2020-10-10:
please！
I am  a Green-hand.
I am willing to learn something about linux.

thundron, 2020-10-13:
@zhang2514yuchi it seems that you linked to another resource instead of the image, but besides that:
did you check that the USB stick works as intended? Does the guide say to use Etcher or perhaps another tool, since you're trying to create a bootable USB?

lurch, 2020-10-13:
@thundron That image-link worked a couple of days ago (ignoring the fact the markdown link syntax was typed in the wrong-way-around), dunno why it's now failing ? (but I can't remember now what it said)
@zhang2514yuchi Perhaps you could provide a like to whatever "tutorial" or  "official document" it is that you're following?

zhang2514yuchi, 2020-10-14:
@lurch 
 [tutorial](https://discourse.ubuntu.com/t/create-a-bootable-usb-stick-on-macos/14016)
 [image](https://cdn.luogu.com.cn/upload/image_hosting/7m3qgj6e.png)
thanks

zhang2514yuchi, 2020-10-14:
@thundron @lurch 
[tutorial](https://discourse.ubuntu.com/t/create-a-bootable-usb-stick-on-macos/14016)
error：
1 failed target
/dev/disk2:Checksum does not match for range [0,2785017855]

gelbal, 2020-10-26:
I believe this is a duplicate of https://github.com/balena-io/etcher/issues/3197.

@zhang2514yuchi, do you get any error running the flashed USB stick?