balena-io/etcher #3320: Etcher reports "user did not grant access"

BertLindeman, 2020-10-11:
- **Etcher version:** 
1.5.109

- **Operating system and architecture:** 
Windows 10 x86_64

- **Image flashed:** 
```https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2020-04-10/snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz```
Downloaded the compressed image and before flashing I unblocked the file (due to being downloaded)
Also extracted the image with the same result.
The same with another image:  
```https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2020-03-23/snapshot-ev3dev-stretch-rpi-generic-2020-03-23.img.xz```

- **Do you see any meaningful error information in the DevTools?** 
(reformatted a bit - set "\n newline" on a newline for better readability (I hope)
```
Version = 1.5.109, Type = nsis
gui.js:56225 Sun Oct 11 2020 15:29:47 GMT+0200 (Midden-Europese zomertijd) Application start ({"packageType":"nsis","version":"1.5.109","sample":0.1,"applicationSessionUuid":"99ed2e01-c944-4509-ad7c-6b3a3e252fca","flashingWorkflowUuid":"bf076ab9-537f-47de-aa06-b2c626354fdd"})
gui.js:48927 Sun Oct 11 2020 15:30:55 GMT+0200 (Midden-Europese zomertijd) Open image selector ({"sample":0.1,"applicationSessionUuid":"99ed2e01-c944-4509-ad7c-6b3a3e252fca","flashingWorkflowUuid":"bf076ab9-537f-47de-aa06-b2c626354fdd"})
gui.js:48927 Sun Oct 11 2020 15:31:07 GMT+0200 (Midden-Europese zomertijd) Select image ({"image":{"isCompressed":true,"name":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img","size":1887436800,"compressedSize":271860216,"isSizeEstimated":false,"hasMBR":true,"partitions":[{"offset":4194304,"size":50331648,"type":11,"index":1},{"offset":54525952,"size":1832910848,"type":131,"index":2}],"extension":"xz","path":"C:\\Lubuntu\\rpi0red\\sdcard\\snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz","logo":false,"blockMap":false},"sample":0.1,"applicationSessionUuid":"99ed2e01-c944-4509-ad7c-6b3a3e252fca","flashingWorkflowUuid":"bf076ab9-537f-47de-aa06-b2c626354fdd"})
gui.js:48927 Sun Oct 11 2020 15:31:24 GMT+0200 (Midden-Europese zomertijd) Toggle drive ({"drive":{"size":15931539456,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"D:\\"}],"description":"TS-RDF5 SD  Transcend USB Device","isUAS":false},"previouslySelected":false,"sample":0.1,"applicationSessionUuid":"99ed2e01-c944-4509-ad7c-6b3a3e252fca","flashingWorkflowUuid":"bf076ab9-537f-47de-aa06-b2c626354fdd"})
gui.js:48927 Sun Oct 11 2020 15:31:36 GMT+0200 (Midden-Europese zomertijd) Flash ({"image":{"size":1887436800,"compressedSize":271860216,"hasMBR":true,"path":"C:\\Lubuntu\\rpi0red\\sdcard\\snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz","partitions":[{"offset":4194304,"size":50331648,"type":11,"index":1},{"offset":54525952,"size":1832910848,"type":131,"index":2}],"isCompressed":true,"name":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img","isSizeEstimated":false,"extension":"xz"},"drives":[{"size":15931539456,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"D:\\"}],"description":"TS-RDF5 SD  Transcend USB Device","isUAS":false}],"driveCount":1,"uuid":"1e57b03e-da98-4a4a-a603-b9a7d6f96a06","status":"started","flashInstanceUuid":"1e57b03e-da98-4a4a-a603-b9a7d6f96a06","unmountOnSuccess":true,"validateWriteOnSuccess":true,"sample":0.1,"applicationSessionUuid":"99ed2e01-c944-4509-ad7c-6b3a3e252fca","flashingWorkflowUuid":"bf076ab9-537f-47de-aa06-b2c626354fdd"})
gui.js:48927 Elevating command: C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\balenaEtcher.exe C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\resources\app\generated\child-writer.js
gui.js:6038  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:48927 Sun Oct 11 2020 15:31:37 GMT+0200 (Midden-Europese zomertijd) SafeWebview loaded ({"event":{"id":1,"url":"https://assets.balena.io/etcher-featured/index.html?borderRight=false&darkBackground=true&etcher-version=1.5.109&api-version=2&optOutAnalytics=false","method":"GET","timestamp":1602423097868.207,"resourceType":"mainFrame","ip":"2606:4700:3035::ac43:9567","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"access-control-allow-origin":["*"],"age":["545"],"cache-control":["max-age=600"],"cf-cache-status":["DYNAMIC"],"cf-ray":["5e08f1491c3ad8f5-AMS"],"cf-request-id":["05b97521af0000d8f5199e8200000001"],"content-encoding":["br"],"content-type":["text/html; charset=utf-8"],"date":["Sun, 11 Oct 2020 13:31:37 GMT"],"expect-ct":["max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\""],"expires":["Sun, 11 Oct 2020 06:22:40 GMT"],"last-modified":["Tue, 18 Aug 2020 09:51:57 GMT"],"nel":["{\"report_to\":\"cf-nel\",\"max_age\":604800}"],"report-to":["{\"endpoints\":[{\"url\":\"https:\\/\\/a.nel.cloudflare.com\\/report?lkg-colo=20&lkg-time=1602423098\"}],\"group\":\"cf-nel\",\"max_age\":604800}"],"server":["cloudflare"],"status":["200"],"vary":["Accept-Encoding"],"via":["1.1 varnish"],"x-cache":["HIT"],"x-cache-hits":["2"],"x-fastly-request-id":["b73586cc665197729346f88e54bc01428aa316ba"],"x-github-request-id":["B468:12A26:3FB1D:565F6:5F82A258"],"x-origin-cache":["HIT"],"x-proxy-cache":["MISS"],"x-served-by":["cache-ams21081-AMS"],"x-timer":["S1602423098.781463,VS0,VE0"]},"referrer":"","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"99ed2e01-c944-4509-ad7c-6b3a3e252fca","flashingWorkflowUuid":"bf076ab9-537f-47de-aa06-b2c626354fdd"})
gui.js:48927 Terminating IPC server
gui.js:48927 Flash results {}
gui.js:48927 Sun Oct 11 2020 15:31:42 GMT+0200 (Midden-Europese zomertijd) Write failed ({"image":{"size":1887436800,"compressedSize":271860216,"hasMBR":true,"path":"C:\\Lubuntu\\rpi0red\\sdcard\\snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz","partitions":[{"offset":4194304,"size":50331648,"type":11,"index":1},{"offset":54525952,"size":1832910848,"type":131,"index":2}],"isCompressed":true,"name":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img","isSizeEstimated":false,"extension":"xz"},"drives":[{"size":15931539456,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"D:\\"}],"description":"TS-RDF5 SD  Transcend USB Device","isUAS":false}],"driveCount":1,"uuid":"1e57b03e-da98-4a4a-a603-b9a7d6f96a06","status":"failed","flashInstanceUuid":"1e57b03e-da98-4a4a-a603-b9a7d6f96a06","unmountOnSuccess":true,"validateWriteOnSuccess":true,"error":{},"sample":0.1,"applicationSessionUuid":"99ed2e01-c944-4509-ad7c-6b3a3e252fca","flashingWorkflowUuid":"bf076ab9-537f-47de-aa06-b2c626354fdd"})
gui.js:48927 {"stack":"Error: User did not grant permission.
\n    at file:///C%3A/Users/bert.grijshuid/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:29095:35
\n    at ChildProcess.exithandler (child_process.js:311:5)
\n    at ChildProcess.emit (events.js:223:5)
\n    at maybeClose (internal/child_process.js:1021:16)
\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","message":"User did not grant permission.","image":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz"}
console.<computed> @ gui.js:48927
logException @ gui.js:56295
flashImageToDrive @ gui.js:79583
async function (async)
flashImageToDrive @ gui.js:79572
tryFlash @ gui.js:79662
callback @ gui.js:79684
ba @ gui.js:42470
ja @ gui.js:42484
ka @ gui.js:42487
oa @ gui.js:42501
lc @ gui.js:43172
jc @ gui.js:43166
mc @ gui.js:43183
sc @ gui.js:43252
Ia @ gui.js:48482
Ma @ gui.js:42580
Rc @ gui.js:43518
id @ gui.js:43487
exports.unstable_runWithPriority @ gui.js:48811
cg @ gui.js:44679
Ga @ gui.js:48470
gd @ gui.js:43477
```

**Some extra info**
Trying to flash the "file from URL" failed the same.

The extracted file from 
```https://oss.jfrog.org/list/oss-snapshot-local/org/ev3dev/brickstrap/2020-04-10/snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz```
was written successfully using 
```HDDGURU Raw Copy Tool\RawCopy.exe```

thundron, 2020-10-13:
@BertLindeman have you got any application/OS configuration that might prevent Etcher to access the source or target? Something like Controlled Folder Access or an antivirus could get in the way of accessing the resources

BertLindeman, 2020-10-14:
> @BertLindeman have you got any application/OS configuration that might prevent Etcher to access the source or target? Something like Controlled Folder Access or an antivirus could get in the way of accessing the resources

@thundron I do not think so. 
Kept all the same:
. same PC
. same USB reader with the same micro sd-card
. same image to be flashed.

Took Etcher 1.5.39 and flashed like I like it. 

Thinking if there was some access / antivirus refusing this, it would also be the case for the older etcher.

Worked like I am used to with Etcher. I like it SO much!  Thanks Balena for Etcher. 

Not sure if this helps, but a large part of the console:

```
Download the React DevTools for a better development experience: https://fb.me/react-devtools
You might need to use a local HTTP server (instead of file://): https://fb.me/react-devtools-faq
C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Warning: The "extend" API will be removed in the upcoming v4.0 release. Use styled(StyledComponent) instead. You can find more information here: https://github.com/styled-components/styled-components/issues/1546
console.(anonymous function) @ C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58
C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.39, Type = nsis
C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Oct 14 2020 09:16:08 GMT+0200 (West-Europa (zomertijd)) Application start ({"packageType":"nsis","version":"1.5.39","applicationSessionUuid":"1af9bf7a-effc-450f-8b20-a3a84e5886ed","sample":0.1})
C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Oct 14 2020 09:16:09 GMT+0200 (West-Europa (zomertijd)) SafeWebview loaded ({"event":{"fromCache":false,"headers":{"Accept":"text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8","Upgrade-Insecure-Requests":"1","User-Agent":"Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) balena-etcher/1.5.39 Chrome/66.0.3359.181 Electron/3.1.9 Safari/537.36"},"id":10,"method":"GET","referrer":"","resourceType":"mainFrame","responseHeaders":{"age":["71133"],"cache-control":["public, max-age=0, must-revalidate"],"content-encoding":["br"],"content-length":["6446"],"content-type":["text/html; charset=UTF-8"],"date":["Tue, 13 Oct 2020 11:30:36 GMT"],"etag":["\"e536574d00183d7f040d0d865889ac1a-ssl-df\""],"link":["</webpack-runtime-af27a6992a13ae8be045.js>; rel=preload; as=script, </styles-ae6d046c35c187471a06.js>; rel=preload; as=script, </framework-3c7631283b89703ace38.js>; rel=preload; as=script, </73ad3401-f1b16c54c781b6334272.js>; rel=preload; as=script, </532a2f07-f8962197ebd575048aef.js>; rel=preload; as=script, </app-64464c10226066ae3025.js>; rel=preload; as=script, </c943faba-db53744371810ee1b6a2.js>; rel=preload; as=script, </af13d906-d19bd41044e43a18d79c.js>; rel=preload; as=script, </ec1189df-cd9eecc4bce7e8eb393f.js>; rel=preload; as=script, </29107295-cc8712978f6051502705.js>; rel=preload; as=script, </a9a7754c-89160c11e5d846782e95.js>; rel=preload; as=script, </4cb05eeb-76399205f5e8cc3a6edf.js>; rel=preload; as=script, </231dbb27-68eda63ea4bd6b7d6c02.js>; rel=preload; as=script, </09701b4e-679cc6225d7a7c57508c.js>; rel=preload; as=script, </8adc965a-0a7d3c2c406e33a9f402.js>; rel=preload; as=script, </42cf64af-75df0380b6cdc8c7c1fc.js>; rel=preload; as=script, </commons-dbf4ce0d9e6b856e6f0f.js>; rel=preload; as=script, </fa5ad257d571af33d737afaa118648b294b3ca41-1483844bce6db0eb141c.js>; rel=preload; as=script, </1f9101bfa6ab6b5c77f1b7191fb4987950aee8e9-d8f1dbc3ea16a0377679.js>; rel=preload; as=script, </component---src-pages-etcher-success-banner-js-06231f4a7d00f7b2d875.js>; rel=preload; as=script, </page-data/app-data.json>; rel=preload; as=fetch; crossorigin, </page-data/etcher/success-banner/page-data.json>; rel=preload; as=fetch; crossorigin"],"referrer-policy":["same-origin"],"server":["Netlify"],"status":["200"],"strict-transport-security":["max-age=31536000"],"vary":["Accept-Encoding"],"x-content-type-options":["nosniff"],"x-frame-options":["DENY"],"x-nf-request-id":["94ed4a00-e242-4e0b-b1b4-900e7cd5304f-33692747"],"x-xss-protection":["1; mode=block"]},"statusCode":200,"statusLine":"HTTP/1.1 200","timestamp":1602659769784.021,"url":"https://www.balena.io/etcher/success-banner/?etcher-version=1.5.39&api-version=2&optOutAnalytics=false"},"applicationSessionUuid":"1af9bf7a-effc-450f-8b20-a3a84e5886ed","flashingWorkflowUuid":"8a50aa4d-c8d4-4168-add3-d0c1cb0991ef","sample":0.1})
C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Oct 14 2020 09:16:14 GMT+0200 (West-Europa (zomertijd)) Open external link ({"url":"https://github.com/resin-io/etcher/blob/master/SUPPORT.md","applicationSessionUuid":"1af9bf7a-effc-450f-8b20-a3a84e5886ed","sample":0.1})
[... cut ...] 49","unmountOnSuccess":true,"validateWriteOnSuccess":true,"applicationSessionUuid":"1af9bf7a-effc-450f-8b20-a3a84e5886ed","flashingWorkflowUuid":"8a50aa4d-c8d4-4168-add3-d0c1cb0991ef","sample":0.1})
C:\Users\bert.grijshuid\AppData\Local\Programs\balena-etcher\resources\app.asar\node_modules\raven-js\dist\raven.js:58 Wed Oct 14 2020 09:22:35 GMT+0200 (West-Europa (zomertijd)) Navigate ({"to":"success","from":"main","applicationSessionUuid":"1af9bf7a-effc-450f-8b20-a3a84e5886ed","sample":0.1})
```

BertLindeman, 2020-10-17:
Ran older portable versions from [etcher releases](https://github.com/balena-io/etcher/releases)
1.5.107 and 1.5.108 gave an error (showing below the 1.5.108 error):

```
Decompressing , 100% at 123.40 MB/s (total 0.00 MB/s) eta in 0s with 0 failed devices
gui.js:48918 Terminating IPC server
gui.js:48918 Flash results {}
gui.js:48918 Sat Oct 17 2020 09:42:17 GMT+0200 (Midden-Europese zomertijd) Write failed ({"image":{"size":1887436800,"compressedSize":271860216,"hasMBR":true,"path":"C:\\Lubuntu\\rpi0red\\sdcard\\snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz","partitions":[{"offset":4194304,"size":50331648,"type":11,"index":1},{"offset":54525952,"size":1832910848,"type":131,"index":2}],"isCompressed":true,"name":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img","isSizeEstimated":false,"extension":"xz"},"drives":[{"size":15931539456,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"D:\\"}],"description":"TS-RDF5 SD  Transcend USB Device","isUAS":false}],"driveCount":1,"uuid":"3b1e9cd4-d7d1-45de-83d5-7a10bc8f69e7","status":"failed","flashInstanceUuid":"3b1e9cd4-d7d1-45de-83d5-7a10bc8f69e7","unmountOnSuccess":true,"validateWriteOnSuccess":true,"error":{"code":-1073741819},"sample":0.1,"applicationSessionUuid":"12babd28-2f76-4cef-845b-b6c7e3268334","flashingWorkflowUuid":"10f3e2dd-504c-48d2-ad93-1b9381a6b849"})
gui.js:48918 {"stack":"Error: Command failed: cmd /c \"C:\\Users\\bert.grijshuid\\AppData\\Local\\Temp\\balena-etcher-electron--20052-l6vxE6inDvrE-.cmd\"\r\n\n    at file:///C%3A/Users/bert.grijshuid/AppData/Local/Temp/1hJiTKxQC5OF9PkZrCupnveh2JT/resources/app/generated/gui.js:29102:33\n    at FSReqCallback.readFileAfterClose [as oncomplete] (internal/fs/read_file_context.js:61:3)","message":"Command failed: cmd /c \"C:\\Users\\bert.grijshuid\\AppData\\Local\\Temp\\balena-etcher-electron--20052-l6vxE6inDvrE-.cmd\"\r\n","code":-1073741819,"image":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz"}
console.<computed> @ gui.js:48918
logException @ gui.js:56286
flashImageToDrive @ gui.js:79574
```

After the error re-tried `Version = 1.5.105, Type = portable` that ran OK before I ran 1.5.107 and 108.
Still runs OKE. 1.5.105 end of log:

```
Verifying 1 device, 100% at 43.10 MB/s (total 43.10 MB/s) eta in 0s with 0 failed devices
gui.js:48816 Finish: 1102053376
gui.js:48816 Terminating IPC server
gui.js:48816 Flash results {results: {…}, cancelled: false}
gui.js:48816 Sat Oct 17 2020 09:52:01 GMT+0200 (Midden-Europese zomertijd) Done ({"image":"C:\\Lubuntu\\rpi0red\\sdcard\\snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz","drives":[{"size":15931539456,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"\\\\.\\PhysicalDrive1","busVersion":"2.0","isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":null,"mountpoints":[{"path":"D:\\"}],"description":"TS-RDF5 SD  Transcend USB Device","isUAS":false}],"driveCount":1,"uuid":"c7aadddd-cd37-412f-b69f-e8b087d659c7","status":"finished","flashInstanceUuid":"c7aadddd-cd37-412f-b69f-e8b087d659c7","unmountOnSuccess":true,"validateWriteOnSuccess":true,"errors":[],"devices":{"failed":0,"successful":1},"bytesWritten":1102053376,"sourceMetadata":{"size":1887436800,"name":"1553e4ac7d8f.tmp","blocks":[{"blocks":[{"offset":0,"length":100663296}]},{"blocks":[{"offset":188743680,"length":973078528}]},{"blocks":[{"offset":1195376640,"length":27262976}]},{"blocks":[{"offset":1262485504,"length":1048576}]}],"blockmappedSize":1102053376},"sample":0.1,"applicationSessionUuid":"0c936e7d-1aac-48b9-9e04-2451c30b13db","flashingWorkflowUuid":"a5f1c274-ba27-41d6-b439-968acd086f5c"})
gui.js:48816 Sat Oct 17 2020 09:52:02 GMT+0200 (Midden-Europese zomertijd) SafeWebview loaded ({"event":{"id":20,"url":"https://www.balena.io/etcher/success-banner/?etcher-version=1.5.105&api-version=2&optOutAnalytics=false","method":"GET","timestamp":1602921122147.5889,"resourceType":"mainFrame","ip":"2a03:b0c0:3:e0::32e:b001","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"age":["147363"],"cache-control":["public, max-age=0, must-revalidate"],"content-encoding":["br"],"content-length":["6444"],"content-type":["text/html; charset=UTF-8"],"date":["Thu, 15 Oct 2020 14:55:57 GMT"],"etag":["\"e4af00da526f6a4ac0485893df07d5e6-ssl-df\""],"link":["</webpack-runtime-6da69eccfb9e5252bee7.js>; rel=preload; as=script, </styles-ae6d046c35c187471a06.js>; rel=preload; as=script, </framework-3c7631283b89703ace38.js>; rel=preload; as=script, </73ad3401-f1b16c54c781b6334272.js>; rel=preload; as=script, </532a2f07-f8962197ebd575048aef.js>; rel=preload; as=script, </app-64464c10226066ae3025.js>; rel=preload; as=script, </c943faba-db53744371810ee1b6a2.js>; rel=preload; as=script, </af13d906-d19bd41044e43a18d79c.js>; rel=preload; as=script, </ec1189df-cd9eecc4bce7e8eb393f.js>; rel=preload; as=script, </29107295-cc8712978f6051502705.js>; rel=preload; as=script, </a9a7754c-89160c11e5d846782e95.js>; rel=preload; as=script, </4cb05eeb-76399205f5e8cc3a6edf.js>; rel=preload; as=script, </231dbb27-68eda63ea4bd6 [... cut ...] s":["nosniff"],"x-frame-options":["DENY"],"x-nf-request-id":["9e18e659-6df5-4c8a-82c9-cdcc771b194a-88559443"],"x-xss-protection":["1; mode=block"]},"referrer":"","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"75889fc2-beab-4ddf-9b05-8baa2612dc5a","flashingWorkflowUuid":"283c7752-8523-4ded-92b1-150e53d98e3e"})
```

So my conclusion (today running only portable versions) would be that this problems starts in 1.5.107.
Could be that the problem in 1.5.109 differs from 1.5.107 and 1.5.108.

thundron, 2020-10-17:
@BertLindeman good job at bisecting the version where it doesn't work! We did in fact change some things related to the permissions when the flashing sub-process is spawned on Windows, the weird thing is that it works for some and doesn't for others. Having a different arch (32/64bit) seemed to have the greatest impact but we tested on both 32bit and 64bit and it worked so there must be other subtle differences which we haven't been able to pinpoint yet

BertLindeman, 2020-10-17:
@thundron Thanks for your response.
Well, I have been testing (mainframe) software for so long....
Windows updates sometimes may show subtle differences.
I will keep the image and micro SD-card for some time so I can test again.

BertLindeman, 2020-11-05:
Re-tried using etcher 1.5.110 portable without success:

```
{"stack":"Error: User did not grant permission.\n    at file:///C%3A/Users/bert.grijshuid/AppData/Local/Temp/1jpMXclGkl78v6lWbk0drArdMMU/resources/app/generated/gui.js:29100:35\n    at ChildProcess.exithandler (child_process.js:311:5)\n    at ChildProcess.emit (events.js:223:5)\n    at maybeClose (internal/child_process.js:1021:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","message":"User did not grant permission.","image":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz"}
console.<computed> @ gui.js:48932
o95242.ingest.sentry.io/api/5504271/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=7118db6e55374aa68877e051c40f4f4c:1 Failed to load resource: the server responded with a status of 403 (Forbidden)
```

shawaj, 2020-11-23:
@BertLindeman FYI v1.5.111 is available now https://www.balena.io/etcher/

BertLindeman, 2020-11-24:
The download of portable v1.5.111 was/is a crime, repeated stalls or it even failed. Tried the download from 09:00 CET.
Probably server problems as my download speed at the moment was 30 Mb/s. Well forget about it.
Testing same file as before (\snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz) flashing onto same sd-card.
Using v1.5.111 this succeeds! 
We need to take in account that in the meantime updates on Win10 have been applied.
Verifying by re-flashing now with v.1.5.110.
Well good news, v1/5/110 still fails, so no Windows update did interfere with this issue.

Last part of the log (v1.5.110):

```
Tue Nov 24 2020 09:44:59 GMT+0100 (Midden-Europese standaardtijd) Write failed ({"image":{"size":1887436800,"compressedSize":271860216,"hasMBR":true,"path":"C:\\Lubuntu\\rpi0red\\sdcard\\snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz","partitions":[{"offset":4194304,"size":50331648,"type":11,"index":1},{"offset":54525952,"size":1832910848,"type":131,"index":2}],"isCompressed":true,"name":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img","isSizeEstimated":false,"extension":"xz"},"drives":[{"size":15931539456,"isVirtual":false,"enumerator":"USBSTOR","logicalBlockSize":512,"raw":"\\\\.\\PhysicalDrive1","error":null,"isReadOnly":false,"displayName":"D:\\, "}],"description":"TS-RDF5 SD  Transcend USB Device","isUAS":false}],"driveCount":1,"uuid":"edb89f32-ae32-4634-8c6c-937878b4b677","status":"failed","flashInstanceUuid":"edb89f32-ae32-4634-8c6c-937878b4b677","unmountOnSuccess":true,"validateWriteOnSuccess":true,"error":{},"sample":0.1,"applicationSessionUuid":"12f388e3-0302-4d6a-8a74-00dc0530d674","flashingWorkflowUuid":"cb761f1b-53f3-4325-875f-d818ea7bc687"})
gui.js:48932 {"stack":"Error: User did not grant permission.
 at 
file:///C%3A/Users/bert.grijshuid/AppData/Local/Temp/1jpMXclGkl78v6lWbk0drArdMMU/resources/app/generated/gui.js:29100:35
    at ChildProcess.exithandler (child_process.js:311:5)
    at ChildProcess.emit (events.js:223:5)
    at maybeClose (internal/child_process.js:1021:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","message":"User did not grant permission.","image":"snapshot-ev3dev-buster-rpi-generic-2020-04-10.img.xz"}
console.<computed> @ gui.js:48932
o95242.ingest.sentry.io/api/5504271/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=7118db6e55374aa68877e051c40f4f4c:1 
Failed to load resource: the server responded with a status of 403 (Forbidden)
```

No idea what change in v1.5.111 resolved this, but this issue may be closed, unless development thinks otherwise ;-)

THANKS!

shawaj, 2020-11-24:
@BertLindeman thanks for the update and glad it is now working for you 👍 

Will leave the closing / further investigation up to @thundron

BertLindeman, 2020-11-24:
> 
> 
> @BertLindeman thanks for the update and glad it is now working for you 👍
> 
> Will leave the closing / further investigation up to @thundron

Good ;-) 
Thanks, Etcher is so easy to use and so important now-a-days for RPi / Arduino, etc. . . .

closed 2020-12-08 by shawaj