balena-io/etcher #3321: Etcher should not fake a system password prompt

luizberti, 2020-10-11:
- **Etcher version:** All since commit 9b82891abb86fc27c3df531fa6e1086192de4f03 from @zvin 
- **Operating system and architecture:** macOS (but possibly Linux as well, I don't it on desktop platforms to validate this)
- **Image flashed:** Does not apply
- **Do you see any meaningful error information in the DevTools?** Does not apply

Before flashing a device, Etcher shows what appears to be a system prompt asking for the user's password, but it is not in fact a system prompt: Etcher's own application is emulating a system prompt and capturing the password itself.

This tricks users, who think they are entering their password into a trusted system prompt, into forfeiting their password to an arbitrary userspace application who can do with it as it pleases, subverting the system's own mechanisms for escalating a process's privilege, so that the application can choose at it's own whim what to escalate, rather than the user.

This is similar to what [Zoom's notoriously shady installer did to bypass escalation prompts](https://www.vmray.com/cyber-security-blog/zoom-macos-installer-analysis-good-apps-behaving-badly), and even though not necessarily indicative of malicious practices, it's terrible practice and a complete betrayal of the user's trust, not to mention a possible malware vector.

Please, stop emulating system prompts and leverage the standard system mechanisms to implement process escalation. The RaspberryPi Imager app is similar in intent to Etcher and as far as I could tell, seems to be using the standard system mechanisms to escalate privilege when it's time to flash (as a matter of fact, it prompts for a much more granular and specific access than full-blown `sudo`), you should do the same.

lurch, 2020-10-11:
See also #3309

luizberti, 2020-10-12:
From the #3309 link I also found https://github.com/balena-io/etcher/issues/2975 and https://github.com/balena-io/etcher/issues/3107 which are related. Based on [this comment](https://github.com/balena-io/etcher/issues/3107#issuecomment-693633149) from the author of #3107, should we leave this issue open and continue tracking this problem through here?

zvin, 2020-10-12:
Etcher never sees the password, we use `sudo --askpass`.
The only code that "sees" the password is https://github.com/balena-io/etcher/blob/master/lib/shared/catalina-sudo/sudo-askpass.osascript.js , it is invoked by `sudo` ( https://github.com/balena-io/etcher/blob/cc08ac9236fad0e98700ab403726116122b821c6/lib/shared/catalina-sudo/sudo.ts#L34 ) and is not part of etcher itself.
Etcher never sees your password.

> Please, stop emulating system prompts and leverage the standard system mechanisms to implement process escalation.

Please read the other issues, this no longer works for writing to `/dev/disk*` starting from Catalina.

> not to mention a possible malware vector.
How ?

I had a quick look at rpi-imager, it uses `authopen`, this could probably work for etcher too but requires some native code https://github.com/raspberrypi/rpi-imager/blob/qml/mac/macfile.cpp#L25 . Pull requests are welcome.

niallrr, 2021-02-06:
I'll look into a way to resolve this. Pretty alarming when you first notice it before coming here.

poohomar, 2021-03-01:
I don't understand how these requests are being closed with no native function in place! Why do the end users have to find a way to use your application, no other macOS application you have to either run as root via terminal or logout/switch user to an admin account to run the application? Why is admin privileges needed to create a bootable usb anyway?

lurch, 2021-03-01:
> Why is admin privileges needed to create a bootable usb anyway?

Because Etcher needs low-level raw block access to the disk, which (from a security point of view) is very different to writing files onto a filesystem.

mcx808, 2021-03-01:
The Raspberry Pi Imager requires no such escalation shenanigans. It would be great to include the same method they use, but of course it could be some limitation of etcher being based on electron.
https://github.com/raspberrypi/rpi-imager

Emulating a system prompt for a users password is a big no-no.

thundron, 2021-03-22:
@poohomar completely agree that there is a better way of doing it *somewhere*, we just haven't got the time to get on this specific issue.
PRs are welcome though, so if anyone of you wants to give it a try, as this is an open-source project, feel free to! :)

defensivedepth, 2021-10-04:
I'll throw my hat into this discussion as well, as the non-system password prompt actually made me a bit concerned that Etcher was doing something nefarious. Googled around a bit and found this issue.

kmbgit, 2023-07-27:
Calling `sudo` on a Mac is the wrong thing to do.  My use account type is "standard" and you need to become an "administrator" to access the low-level disk device.  As others have mentioned, the RaspberryPi Imager manages this feat.  You don't need to become root do do it, you just need to ask the right question.

bradleypeabody, 2023-08-14:
Just another vote for fixing this - downloaded and tried Etcher today and immediately went searching to see if this is spyware.  Password prompts should always be from the system, period.  An app-specific dialog asking for a password is basically a big sign that says "I'm here to steal your information and do something, just checking to see if you are foolish enough to fall for it".

It seems like this wasn't done maliciously but, yeah, just another opinion here reiterating the point - this is a really bad and scary user experience.

thilaire, 2023-10-18:
I totally agree with Bradleyoeabody.
I am here just because I found very strange that a program ask for my password, and not ask the system for access.
I will try to see if there is an alternative (ie another program that do not ask my password, but rather ask access to the system).

rwalle, 2024-01-18:
> Just another vote for fixing this - downloaded and tried Etcher today and immediately went searching to see if this is spyware. Password prompts should always be from the system, period. An app-specific dialog asking for a password is basically a big sign that says "I'm here to steal your information and do something bad with it, just checking to see if you are foolish enough to fall for it".
> 
> It seems like this wasn't done maliciously but, yeah, just another opinion here reiterating the point - this is a really bad and scary user experience. Etcher needs _access_, not my password. It's 2023, a lot more users know the difference these days.

I find it puzzling that an application with such bad security practice and dubious privacy handling is often the recommended tool for flashing images. Well, I guess there often isn't a better alternative.

Golffies, 2024-04-10:
I hope that a patch will be proposed to bring Etcher into line with good practice. Continuing to use Etcher as it works today is like living in the world of Dune, permanently wearing a heart valve on your chest: everything's fine as long as nobody pulls on it.

Users can't be expected to check the source code for potential malicious changes every time they accept an update from Etcher. If we want to be consistent, we must uninstall Etcher on macOS as long as the application persists in asking the user for his password rather than requesting privileges to be raised from the system. 

This comment posted on #4026 states that the Raspberry Pi Imager application for macOS obtains the right to write blocks to a device directly from the OS, and not by collecting the password directly from the user.

> Alternatively, download the Raspberry Pi Imager from https://www.raspberrypi.com/software/ - it manages to ask for an admin username/password so it must be possible.

mcx808, 2024-04-10:
> Locking the conversation as it's not our priority right now to beautify and/or normalize the dialogue to a system one, as it involves solving so many other parallel issues; we already made clear that we're more than welcome to receive PRs regarding the subject though, so if anybody feels like doing it, that would be awesome!

This is a shame, and I would hope projects would reconsider this stance after the recent XZ compromise. I get there's no guarantees with FOSS but sure this attitude will reflect on Balenas commercial offerings as well.

dfunckt, 2024-04-10:
You’re right and it’s embarrassing but it’s not for lack of trying to resolve this. Digging into the problem the only way forward appears to be to refactor a huge chunk of Etcher into what Apple calls XPC services, which is not a project we can take on right now.

MarvinLeeSchroeder, 2024-11-20:
It's been four years since this issue was first reported, and it still hasn't been resolved. While I truly appreciate Balena and the fact that it's free to use, I believe addressing the password prompt behavior should be prioritized. I've used numerous other applications and have never encountered this issue and immediately was wondering if the software is actually legit.

AutomationD, 2024-11-28:
I went with a regular `dd` instead

Golffies, 2024-11-28:
_+1_

Here is the command line that could be used to write an ISO image to a usb memory stick. Tested on MacOS using the default dd application:

`sudo dd if=/Users/<depends_on_you>/Downloads/<image_name>.iso of=/dev/<removable_stick_name> bs=1m conv=noerror,sync status=progress`

- if= path and name of the ISO image file
- of= path and name of the USB stick to write to
- bs= block size defined at 1024KB (the best value according to my benchmarks; writing is very slow at the default value of 512 bits)
- conv= obscure options that, according to online readings, are recommended to be enabled for this very purpose; comments are always welcome
- status= will display a progress bar in the console

The ISO image that is burned onto the USB memory stick will then be exactly the same bit by bit as the one that Etcher would have written.

On MacOS, to find out the name of the USB stick 'as a raw block device', use the following command:

`diskutil list`

Before writing to the USB memory stick with dd, it is mandatory to unmount it with the following command:

`sudo diskutil umount /Volumes/<partition_name_on_your_stick>`

To find out the name of the USB stick 'as a volume', use the following command:

`ls -l /Volumes/`

In practice, Etcher is a great tool because it does all this through a GUI that makes it impossible to make a mistake, compared to the many steps and potential question marks an end user has to go through to do the same thing with dd. Both end up working exactly the same.

lurch, 2024-11-28:
> Both end up working exactly the same.

Apart from the fact that it's much simpler and safer, the other advantage of Etcher (and also Raspberry Pi Imager) over `dd` is that it verifies the image was written successfully, in case you have a broken / faulty / fake-capacity USB drive.

rwalle, 2024-11-28:
Just to throw some ideas out there

> In practice, Etcher is a great tool because it does all this through a GUI that makes it impossible to make a mistake, compared to the many steps and potential question marks an end user has to go through to do the same thing with dd. Both end up working exactly the same.

This does not seem hard to do. How much work would it take to someone to create a command line based tool that provides the core features of Etcher along with many of the niceties, without ads, tracking and horrible security practice? You know, you can still have interactive command line tool. Another benefit is that you don't need to bundle a Chrome with a tool that flashes USB drives -- CLI tool with some colors is actually quite usable.

crusy, 2025-04-02:
I just downloaded it and uninstalled it 5 minutes later – no idea what my password is, never needed it. If I had to guess I did not even set a password (but balena does not accept empty input, so maybe I'm wrong), I have an admin account for that. But "of course" balena does not ask for an account (like user _and_ password), but just for a password. So I cannot use it.

ShadSterling, 2025-06-29:
The prompt to "type your password" for sudo is especially misleading for users who have [enabled the root user](https://support.apple.com/en-us/102367) and need to enter the root password.

Wasted an hour or two trying to figure out why the supposedly best app for this never did anything but ask for my password over and over until it timed itself out, only to find out the prompt lied about which password was needed

elsiehupp, 2025-10-06:
@mcx808 mentions [`rpi-imager`](https://github.com/raspberrypi/rpi-imager), and I dug up [the function involved](https://github.com/raspberrypi/rpi-imager/blob/46a9f9aeb81f89f31d613f7c9f66504537fcaf64/src/mac/macfile.cpp#L26-L127):

```cpp
MacFile::authOpenResult MacFile::authOpen(const QByteArray &filename)
{
    int fd = -1;

    QByteArray right = "sys.openfile.readwrite."+filename;
    AuthorizationItem item = {right, 0, nullptr, 0};
    AuthorizationRights rights = {1, &item};
    AuthorizationFlags flags = kAuthorizationFlagInteractionAllowed |
            kAuthorizationFlagExtendRights |
            kAuthorizationFlagPreAuthorize;
    AuthorizationRef authRef;
    if (AuthorizationCreate(&rights, nullptr, flags, &authRef) != 0)
        return authOpenCancelled;

    AuthorizationExternalForm externalForm;
    if (AuthorizationMakeExternalForm(authRef, &externalForm) != 0)
    {
        AuthorizationFree(authRef, 0);
        return authOpenError;
    }

    const char *cmd = "/usr/libexec/authopen";
    QByteArray mode = QByteArray::number(O_RDWR);
    int pipe[2];
    int stdinpipe[2];
    ::socketpair(AF_UNIX, SOCK_STREAM, 0, pipe);
    ::pipe(stdinpipe);
    pid_t pid = ::fork();
    if (pid == 0)
    {
        // child
        ::close(pipe[0]);
        ::close(stdinpipe[1]);
        ::dup2(pipe[1], STDOUT_FILENO);
        ::dup2(stdinpipe[0], STDIN_FILENO);
        ::execl(cmd, cmd, "-stdoutpipe", "-extauth", "-o", mode.data(), filename.data(), NULL);
        ::exit(-1);
    }
    else
    {
        ::close(pipe[1]);
        ::close(stdinpipe[0]);
        ::write(stdinpipe[1], externalForm.bytes, sizeof(externalForm.bytes));
        ::close(stdinpipe[1]);

        const size_t bufSize = CMSG_SPACE(sizeof(int));
        char buf[bufSize];
        struct iovec io_vec[1];
        io_vec[0].iov_base = buf;
        io_vec[0].iov_len = bufSize;
        const size_t cmsgSize = CMSG_SPACE(sizeof(int));
        char cmsg[cmsgSize];

        struct msghdr msg = {0};
        msg.msg_iov = io_vec;
        msg.msg_iovlen = 1;
        msg.msg_control = cmsg;
        msg.msg_controllen = cmsgSize;

        ssize_t size;
        do {
            size = recvmsg(pipe[0], &msg, 0);
        } while (size == -1 && errno == EINTR);

        qDebug() << "RECEIVED SIZE:" << size;

        if (size > 0) {
            struct cmsghdr *chdr = CMSG_FIRSTHDR(&msg);
            if (chdr && chdr->cmsg_type == SCM_RIGHTS) {
                qDebug() << "SCMRIGHTS";
                fd = *( (int*) (CMSG_DATA(chdr)) );
            }
            else
            {
                qDebug() << "NOT SCMRIGHTS";
            }
        }

        pid_t wpid;
        int status;

        do {
            wpid = ::waitpid(pid, &status, 0);
        } while (wpid == -1 && errno == EINTR);

        if (wpid == -1)
        {
            qDebug() << "waitpid() failed executing authopen";
            return authOpenError;
        }
        if (WEXITSTATUS(status))
        {
            qDebug() << "authopen returned failure code" << WEXITSTATUS(status);
            return authOpenError;
        }

        qDebug() << "fd received:" << fd;
    }
    AuthorizationFree(authRef, 0);

    return open(fd, QIODevice::ReadWrite | QIODevice::ExistingOnly | QIODevice::Unbuffered, QFileDevice::AutoCloseHandle) ? authOpenSuccess : authOpenError;
}
```

`rpi-imager` is written in Qt, and this function is written in C++ in order to be able to call Apple's headers via Objective-C++. Most of the relevant functionality, e.g. [`AuthorizationCreate`](https://developer.apple.com/documentation/security/authorizationcreate(_:_:_:_:)), etc., comes from `security/Authorization.h`, i.e. [Authorization Services](https://developer.apple.com/documentation/security/authorization-services).

(I don't know how one usually accesses Apple's Objective-C++ headers via TypeScript. I assume there's some sort of shim?)