balena-io/etcher #3323: flash disk USB drive is locked in Windows 10 - readonly error in linux

cpm9132, 2020-10-14:
I want format my flash disk USB 8GB

- **Etcher version: 1.5.109** 
- **Operating system and architecture: Windows 10 version 1903, Intel core i7 930, and so in Linux Opensuse 15.1 Leap (x86_64)** 
- **Image flashed: 
[image]

** 
- **Do you see any meaningful error information in the DevTools? I don't use it.**

I can't select the checkbox for that target (PhysicalDrive2 8.18GB) ** 

diskpart in Windows Powershell shows similar read-only error.

Same error in linux opensuse Leap 15.1: fdisk read-only filesystem error. I changed the state using hdparam -R0 /dev/sdX, but the error remains. 

I can't format or clean the flash disk USB in both operating systems.

The flash disk USB don't have any lock or switch in the hardware.

shawaj, 2020-12-08:
@cpm9132  This sounds like an issue with your USB stick.

If the operating system is seeing it as locked out read only there's nothing we can really do.

Perhaps this might be helpful - https://www.google.com/amp/s/www.easeus.com/amp/partition-master/format-a-write-protected-usb-flash-drive.html

closed 2020-12-08 by shawaj

maxmod, 2021-05-19:
the best and easy solution use https://formatusb.com/download
    1. ✅ Restore USB Drive Back To Original Full Capacity.
    2. ✅ Check Bad Block in USB Flash Drive.
    3. ✅ You Need to Format the Disk in Drive" error message
    4. ✅ Fix Corrupted Usb drive.
    5. ✅ Fix USB Drive Detected But Not Accessible.
    6. ✅ Format USB drive to Fat32/NTFS/UDF/exFAT/ext2/ext3
    7. ✅ Remove U3 Partition from USB stick .
    8. ✅ Win32 Disk Imager tool to write the IMG to the device only few mb usable
    9. ✅ Fix USB drive unusable, unformattable, and reporting 0 bytes capacity.
    10. ✅ Fix cannot format USB Flash drive in windows
    11. ✅ Windows was unable to complete the format,You need to format before use.
    12. ✅ Fix USB flash drive turned to few MB,GB after writing an ISO image onto it.
    13. ✅ Delete CDFS volume from USB drive in windows to fix USB drive to get
    its original size back.
    14. ✅ Fix Space Full but empty drive USB flash drive says it has all of space
    used but inside all the folders are empty.
    15. ✅ Fix USB Drive Not Showing Up in Windows 7/8/10 due to USB driver issues,
    Drive letter, conflicts File system errors.
    16. ✅ When you get a "The disk is write protected" error whileformatting the
    pendrives, memory sticks,external hard drive or SD Memory Card.
    17. ✅ If your USB drive is shown with a wrong size in PC, due to improper formatting
    of the USB drive. Instead of seeing the actual 16GB,8GB,4GB or whatever the size of your
    USB drive is, you’ll see something like a couple hundred MBs available on it.