balena-io/etcher #3324: etcher is cancer slowedst i have used and most irritating 1 click software on the planet

belenaetcheriscancer, 2020-10-15:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 
this software is fucking uselesss and just corrupted 2 drives while cloning one YOU FUCKING CORRUPTED SOMETHING YOU SHOULD ONLY HAVE READ permission FOR thanks for wasting the last 2 days of my l;ife by destroying my efforts your whole team can suck my shite through an irradiated lead pipe

belenaetcheriscancer, 2020-10-15:
genuinely dont use this software  i would have saved more time copying the fucking binary over file by file by fucking hand honestly i hate your software with so much passion i would fucking donkey slap your staff

Lloir, 2020-10-15:
How would it even corrupt the others if it's just writing to whatever drive you selected?

belenaetcheriscancer, 2020-10-15:
thats exactly why this software is cancer it shouldnt have write permissions for the original drive that it is cloning from yet it does for some stuuuuupid reason and it caused me to loose a heavily modified version of raspian that i was up 3 days customising and building to make distributable for octoprint that would have integrated alexa control for octoprint while hosting 3 instances and much much more for home automation but thats dead now i have to restart and i really dont have the patience now maybe next month as this cost me days over a complete oversight on a 1 click software , when 2 of 3 buttons in your software dont work you know you shouldnt be programming

belenaetcheriscancer, 2020-10-15:
and its likely they didnt even copy the files they cut them from the donor drive and didnt paste them properly leaving the donor bootless and the clone useless

belenaetcheriscancer, 2020-10-15:
i honestly wouldnt shit in a landfill if i knew there was a usb with belena on it near me i would eat my own shit before ever using this software again

JamesH65, 2020-10-15:
What you are writing makes no sense whatsoever.  Donor drive? What on earth is that? You know Etcher doesn't do "file transfers", it copies an image file over an SD card - it only reads from the source file?

I have no idea how you have managed to stuff things up quite so comprehensively as it sounds impossible to me, so it's impressive work.  A prime example of no matter how much you idiot proof your software, someone, somewhere, will break it.

I have no link with the etcher people, except I used it for about 5 years without ever having a single problem. YMMV. Clearly.

closed 2020-10-16 by thundron

thundron, 2020-10-16:
FWIW you might want to re-check not only your manners when interacting with people you don't even know but also whatever it is that you did with your source drive *outside* of Etcher.
Etcher **only** writes to the target by reading from the source. If there's a bug in that part we'll check it and we surely understand the frustration, but it's still not a reason to be disrespectful.