balena-io/etcher #3327: Unable to confirm "sha512sum" for "balenaEtcher-1.5.109-x64.AppImage"

NGenetzky, 2020-10-17:
- 1.5.109
- Linux AppImage

Unless I misunderstand the purpose of this file, it would appear you have a bad checksum. This seems like a security issue.

```
└─>sha512sum /tmp/balenaEtcher-1.5.109-x64.AppImage
56ccf88ad6e6949598903df452bb8bb83036a5f7c5b81cc1bbadfc0915d3ea37a68d005df98aed85ca2d2f53958ee7aec4817663c2dfc158a90a88a7d2978e5d  //tmp/balenaEtcher-1.5.109-x64.AppImage
└─>cat /tmp/latest-linux.yml 
version: 1.5.109
files:
  - url: balenaEtcher-1.5.109-x64.AppImage
    sha512: Vsz4itbmlJWYkD30UruLuDA2pffFuBzBu638CRXT6jemjQBd+YrthcotL1OVjueuxIF2Y8LfwVipCoin0peOXQ==
    size: 89631344
    blockMapSize: 92804
path: balenaEtcher-1.5.109-x64.AppImage
sha512: Vsz4itbmlJWYkD30UruLuDA2pffFuBzBu638CRXT6jemjQBd+YrthcotL1OVjueuxIF2Y8LfwVipCoin0peOXQ==
releaseDate: 2020-09-11T12:52:49.262Z
stagingPercentage: 100
```

lurch, 2020-10-17:
Duplicate of #3219 ?

shawaj, 2020-12-13:
See also #3341 and #2826

NGenetzky, 2020-12-14:
Yes, this is a very, very common issue that stems from the confusing way they generate checksums. I will close the issue, but it shocks me that so many tickets have been created about it and yet they don't change it to a more standard format.

closed 2020-12-14 by NGenetzky