balena-io/etcher #3328: stuck on W10 on Starting...

firexx, 2020-10-19:
- **Etcher version:** 1.5.109
- **Operating system and architecture:** Win 10 pro
- **Image flashed:** *.wic
- **Do you see any meaningful error information in the DevTools?** no

[image]

dragonDScript, 2020-10-24:
Have you tried to reload the window/app and try again? Do you see any uncaught promise in the devtools? Have you tried to start as Admin?

firexx, 2020-10-24:
I tried it more times, and also as admin.

dragonDScript, 2020-10-25:
Ok, so these programs fail sometimes.
Try to use dd

BarryOCathain, 2020-10-30:
For info, I was stuck on this issue as well. Ran as admin, and it worked first day.

1435thundercloud, 2021-01-20:
1. hangs for ever lol so i exacted the portable 1.5.115 into a folder. 
2. then portable 1.5.45 into another 
3. 3.then copied all api.dlls from the portable 1.5.45 ver into the 1.5.115 directory.
4. now 1.5.115 starts instantly.

closed 2023-07-12 by dfunckt