balena-io/etcher #3329: User name has hyphen.

Roboblunder, 2020-10-19:
- **1.5.109* 
- **Windows 10:** 
- **2017-07-05-raspbian-jessie.img** 
- **There is a problem with my machine name I believe.  Can't change the name** 

The program is definitely there but I don't think the installation software recognized the computer name which has a hyphen in it.

Here is the error received: ENOENT:no such file or directory, open 'C:\Users\RANDY~1\AppData\Local\Temp\balena-ether-electron--14896-LnYvezNajLuW-.cmd'

I believe the "RANDY~1" is the issue as the username not fully listed here has a Hyphen in it and cannot be changed.  I had a similar problem with Trible Sketchup but they were able to fix it.

Roboblunder, 2020-10-20:
Correction it's not a hyphen it's a apostrophe in the username. The username is similar to "Jerry's Machine".  Names have been changed to protect info. The apostrophe is the thing that seems to be making the flash program confused on the file or directory.

lurch, 2020-10-20:
Sounds similar to #2803 ...

zvin, 2020-11-23:
@Roboblunder can you please try 1.5.111 ?

shawaj, 2020-12-08:
@Roboblunder did you manage to try this out?