balena-io/etcher #3330: Can't write Ubuntu 20.10 for RPi to SDcard

ednl, 2020-10-24:
- **Etcher version:** 1.5.109
- **Operating system and architecture:** MacOS 10.15.7 (x86)
- **Image flashed:** ubuntu-20.10-preinstalled-desktop-arm64+raspi.img.xz _and_ ubuntu-20.10-preinstalled-desktop-arm64+raspi.img
- **Do you see any meaningful error information in the DevTools?** No. I didn't know about DevTools.

Trying to write either the compressed or uncompressed image to a 64 GB SDcard via the built-in SDXC card writer on an iMac, it immediately fails after the password prompt. No particular error specified. I verified the image integrity. The "Raspberry Pi Imager" app does write the image to SDcard error-free.

shawaj, 2020-12-03:
@ednl check #3358 and also try this version - https://github.com/balena-io/etcher/releases/download/untagged-1e30b3a0b27ec01ea938/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-03:
@ednl here is the correct link - https://github.com/balena-io/etcher/releases/download/v1.5.112/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-08:
@ednl assuming the thumbs up means v1.5.112 is working for you?

Will close this issue but feel free to reply back if you are still having issues

closed 2020-12-08 by shawaj