balena-io/etcher #3331: Cannot write Ubuntu on usb stick

Jakub41, 2020-10-24:
- **Etcher version:** 1.5.109 (1.5.109)
- **Operating system and architecture:** Mac Os Catalina 10.15.7 (19H2)10.15.7 (19H2)
- **Image flashed:** Ubuntu 20.04.1 LTS
- **Do you see any meaningful error information in the DevTools?** 

Yes I'm including an error screenshot
[image]

Hello,

I'm unable to write Ubuntu on a USB stick getting an error as something went wrong.

As above included the console error.

I already tried the following solution:

```
Workaround:
Moving your image inside balenaEtcher.app seems to do the trick.
You can do this by selecting "Show Package Contents" on balenaEtcher.app and dragging your image inside of it. Then you can select your image as usual from Finder and the flashing will go ahead. Make sure you select the image inside balenaEtcher.app
```

The ISO image was downloaded fresh from the Ubuntu site a few times so I believe is not a corrupted ISO image. 

Please let me know any other solution which I may try to resolve this situation.

Thank you

thundron, 2020-10-26:
@Jakub41 in the logs I see the error `EBUSY`, which means the resource (in this case the target) is not free for Etcher to use.
If there's anything claiming the resource please close it - it could also be a message from macOS saying that it can't read any partition - and then try again after a couple seconds.

Jakub41, 2020-10-26:
Thank you I resolved using the app with sudo credentials

closed 2020-10-26 by Jakub41