balena-io/etcher #3332: Balena Etcher can't access USB and says it's locked ChromeOS

dragonDScript, 2020-10-24:
- **Etcher version:** 1.5.109
- **Operating system and architecture:** ChromeOS x64 Running Linux
- **Image flashed:** Ubuntu 20.04.1 LTS
- **Do you see any meaningful error information in the DevTools?** No 

Ok, so I have a chromebook and my main PC with Windows was corrupted, so I want to install it Linux. But when trying to select a device, it says that my USB is "locked" like the following image:
[image]
Linux has access to the image, like you see in the following:
[image]
What can I do?
Thank you

dragonDScript, 2020-10-24:
Sorry if you've seen typos, I don't talk a lot of english. Please correct me if you see anything!

lurch, 2020-10-24:
See #1643 and https://github.com/raspberrypi/rpi-imager/issues/92

dragonDScript, 2020-10-25:
Ok, so I've fixed this by renaming hte .iso to .bin and by using the default Chrome OS program to to this. Thank you.

closed 2020-10-25 by dragonDScript

Kentrelle, 2021-10-20:
I also have the same problem. Balenaetcher can only read the USB flash drive and cannot find the USB device.in chromebook