balena-io/etcher #3335: Users would like Etcher releases for iPadOS / iOS / web app

pdcastro, 2020-10-27:
Originally reported at:
https://forums.balena.io/t/webapp-ipad-os-support/203142

> _I am trying to move away from a Mac/Windows based environment for work (it support) and the solution that makes the most (almost) sense to me is running my iPad Pro 2020 as a daily driver. The only drawback is the ability to create multiple Bootable drives On The Fly and I don’t see a solution supported by iPadOS. [...]_
> _I typically will write several different variants of OS X on a weekly basis. I run a repair shop and installing OS X from scratch is a common thing. [...]_

> _+1! It would be great to see Etcher on iPadOS/iOS devices; this is something I’d like for my workflow as well, but I think as (at)crbn60 mentions there is no facility to gain access to storage devices connected via USB at the low level required. At least not last time I checked. Hopefully one of the lovely Etcher team will chime in with more technical info_

thundron, 2020-10-28:
This is definitely beyond the scope of Etcher for now, even more so because there's currently no way to make Electron run on iOS/Android so we would need to re-build basically everything to take into account other frameworks.

closed 2020-10-28 by thundron

deanlyoung, 2020-11-07:
On a related note, what's the strategy to handle ARM-based Macs?