balena-io/etcher #3336: Installation takes excessively long or does not complete

Jestopher-BTC, 2020-10-28:
- **Etcher version: 1.5.109** 
- **Operating system and architecture: Windows 10** 
- **Image flashed: N/A** 
- **Do you see any meaningful error information in the DevTools? N/A** 

The installation process appears to get hung up around the 33% mark and is not completing, nor progressing after an hour on a relatively speedy machine. 
Prior to this issue, I had an older version that I tried to open, but the shortcut couldn't find the executable file. I attempted to uninstall the older version, but Windows 10 couldn't locate the uninstall file.

lurch, 2020-10-28:
Perhaps you need a new SD card? :shrug:

Jestopher-BTC, 2020-10-28:
> Perhaps you need a new SD card? 🤷

I'm just trying to install Balena etcher itself here, not flash an SD card

lurch, 2020-10-29:
I misunderstood, sorry! By "installation process" I thought you meant "installing" an image onto an SD card.

paulknocke, 2020-10-31:
This Etcher works on another Window10-Computer. But visits on another computer are no solution for me.
Where to find an older version of Etcher for a test? My be sufficient to go further?

lurch, 2020-11-01:
> Where to find an older version of Etcher for a test?

https://github.com/balena-io/etcher/releases