balena-io/etcher #3338: Can't write on macOS 19H2

charlesrocket, 2020-10-29:
- **Etcher version:** 1.5.109
- **Operating system and architecture:** macOS 10.15.7
- **Image flashed:** debian-10.6.0-amd64-xfce-CD-1.iso
- **Do you see any meaningful error information in the DevTools?** 

```
persists","report":false},"sample":0.1,"applicationSessionUuid":"0eb08de9-33f0-4909-9ed8-3b5833e52739","flashingWorkflowUuid":"fc96994a-6e27-402a-b588-80f6d540f8ff"})
gui.js:48916 {"stack":"Error: The writer process ended unexpectedly\n    at createError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:63120:23)\n    at createUserError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:63131:16)\n    at Server.<anonymous> (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:79338:28)","message":"The writer process ended unexpectedly","description":"Please try again, and contact the Etcher team if the problem persists","report":false,"image":"debian-10.6.0-amd64-xfce-CD-1.iso"}
```

```
{"size":728760320,"name":"debian-10.6.0-amd64-xfce-CD-1.iso","hasMBR":true,"partitions":[{"offset":3715072,"size":2899968,"type":239,"index":1}],"extension":"iso","path":"/Users/rocket/Downloads/debian-10.6.0-amd64-xfce-CD-1.iso"},"drives":[{"size":32086425600,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/PEG2@1,2/UPSB@0/DSB1@1/UPS0@0/pci-bridge@2/pci8086,15f0@0","mountpoints":[],"description":"Samsung Flash Drive Media","isUAS":null}],"driveCount":1,"uuid":"496ba02d-f908-4323-9efb-4ff1c43b3656","status":"failed","flashInstanceUuid":"496ba02d-f908-4323-9efb-4ff1c43b3656","unmountOnSuccess":true,"validateWriteOnSuccess":true,"error":{"description":"Please try again, and contact the Etcher team if the problem persists","report":false},"sample":0.1,"applicationSessionUuid":"0eb08de9-33f0-4909-9ed8-3b5833e52739","flashingWorkflowUuid":"fc96994a-6e27-402a-b588-80f6d540f8ff"})
```

shawaj, 2020-12-03:
@charlesrocket try this version - https://github.com/balena-io/etcher/releases/download/untagged-1e30b3a0b27ec01ea938/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-13:
@charlesrocket haven't heard back in 10 days so assuming you got this sorted. Will close the issue now.

Feel free to reply back if you don't agree or are still having issues with the latest version.

closed 2020-12-13 by shawaj