balena-io/etcher #3339: Error flashing Ubuntu 18.04 - "The writer process ended unexpectedly"

Sladerix, 2020-11-02:
- **Etcher version:** 
1.5.109
- **Operating system and architecture:** 
macOS Catalina last version
- **Image flashed:** 
ubuntu 18.04
- **Do you see any meaningful error information in the DevTools?** 
gui.js:48916 {"stack":"Error: The writer process ended unexpectedly\n    at createError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:63120:23)\n    at createUserError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:63131:16)\n    at Server.<anonymous> (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:79338:28)","message":"The writer process ended unexpectedly","description":"Please try again, and contact the Etcher team if the problem persists","report":false,"image":"ubuntu-18.04-desktop-amd64.iso"}

nmcc1212, 2020-11-10:
if this isn't already resolved I had the same issue today but when comparing it to the SHA256 checksum found it was corrupt. I just redownloaded the ISO and all was good 
[image]

Sladerix, 2020-11-10:
> if this isn't already resolved I had the same issue today but when comparing it to the SHA256 checksum found it was corrupt. I just redownloaded the ISO and all was good
> [image]

Hello, actually I tried it a week later on another pc, redownloading the .iso on another OS (windows Instead of Mac), and the same, balena crashes

rhymeswithmogul, 2020-11-21:
I am also having the same issue, but with an Ubuntu 20.10 ISO.  I was having the same issue on both macOS Catalina 10.5.7, as well as macOS Big Sur 11.0.1 (Intel).

nmcc1212, 2020-11-21:
is it 64bit?

if so follow this

"Run this command in your terminal in the directory the iso was downloaded to verify the SHA256 checksum:
echo "3ef833828009fb69d5c584f3701d6946f89fa304757b7947e792f9491caa270e *ubuntu-20.10-desktop-amd64.iso" | shasum -a 256 --check"

^ from https://ubuntu.com/download/desktop/thank-you?version=20.10&architecture=amd64

Sladerix, 2020-11-21:
> is it 64bit?
>
> if so follow this
>
>
> ^ from https://ubuntu.com/download/desktop/thank-you?version=20.10&architecture=amd64

Thanks.. in the end I solved by using a previous version of balena

nmcc1212, 2020-11-21:
what an odd issue.

shawaj, 2020-12-03:
take a look at #3358 and try this version - https://github.com/balena-io/etcher/releases/download/untagged-1e30b3a0b27ec01ea938/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-03:
@nmcc1212 @Sladerix @rhymeswithmogul here is the correct link - https://github.com/balena-io/etcher/releases/download/v1.5.112/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-13:
@nmcc1212 @Sladerix @rhymeswithmogul haven't heard back in 10 days so closing this issue. Please feel free to reply back if you don't agree or are still having problems

closed 2020-12-13 by shawaj