balena-io/etcher #3342: RPM Error on Fedora 33

silvernode, 2020-11-02:
- **1.5.109-1:** 
- **Fedora 33/x86_64:** 

**Error Message:**
```
file /usr/lib/.build-id/1b/46d89702261ec7350b37d9c294e9cf30056cec from install of balena-etcher-electron-1.5.109-1.x86_64 conflicts with file from package zettlr-1.7.5-1.x86_64
```

silvernode, 2020-11-02:
Noticed I have a zettlr rpm in my downloads from some point in history, and I don't remember installing it. I removed it from my system and Etcher installed fine. I'll leave this issue up for a bit just to see if anyone has some insight or comments.

lurch, 2020-11-03:
Looks like this is a very similar problem to #3051 (with https://github.com/jordansissel/fpm/issues/1503 possibly being the underlying cause).
See also https://github.com/Zettlr/Zettlr/issues/1321