balena-io/etcher #3344: I am not able to unistall

Thegamer24-7, 2020-11-06:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

shawaj, 2020-11-23:
Have you tried uninstalling from [the windows uninstall or remove programs dialogue](https://support.microsoft.com/en-us/windows/uninstall-or-remove-apps-and-programs-in-windows-10-4b55f974-2cc6-2d2b-d092-5905080eaf98#:~:text=In%20the%20search%20box%20on,the%20directions%20on%20the%20screen.)

If you are using the "portable" version of Etcher then you can just delete the file - as it is not "installed" anywhere.

shawaj, 2020-12-08:
@Thegamer24-7 not heard back in 2 weeks so closing now.

Feel free to reply if you're still having issues.

closed 2020-12-08 by shawaj