balena-io/etcher #3346: Nvidia Jetson Nano image writing issues

tomgalletti, 2020-11-06:
Hi, I've got a Jetson Nano so I'm trying to write the OS image to an SD card. Keep getting the same error. Tried multiple cards and card readers (which work fine with Raspbian on the Raspberry Pi image writer).

- **Etcher version: 1.5.110** 
- **OS: Windows 10 64x** 
- **Image: Jetson Nano OS** 
Error being shown is copied below:

gui.js:48932 {"stack":"Error: User did not grant permission.\n    at file:///C%3A/Users/Rosie/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:29100:35\n    at ChildProcess.exithandler (child_process.js:311:5)\n    at ChildProcess.emit (events.js:223:5)\n    at maybeClose (internal/child_process.js:1021:16)\n    at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)","message":"User did not grant permission.","image":"jetson-nano-4gb-jp441-image.zip.zip"}

Any help appreciated.

Cheers

thundron, 2020-11-06:
@tomgalletti have you got any OS config/application that might prevent Etcher from accessing files/folders/block devices? e.g. on Windows there's Controlled Folder Access for which you might need an exception, or some security software like antiviruses that see Etcher trying to access those resources.
Running as Admin might also help, but I'd advise to check the above first

mattrosenthaltx, 2020-11-08:
Same problem.

jsimmonds2, 2020-11-08:
Same problem on Windows 10 x64.  Any local IMG file or URL.  Seems to be a change around approving elevation (popup).
Etcher v1.5.109 worked without any problems, so I re-installed that version ( Ok ).

mattrosenthaltx, 2020-11-08:
> Etcher v1.5.109

Where can I get older version?

thundron, 2020-11-08:
@mattrosenthaltx https://github.com/balena-io/etcher/tags
@jsimmonds2 thank you for reporting, we're investigating the issue for v1.5.110

thundron, 2020-11-08:
@mattrosenthaltx could you please try running as administrator one last time with v1.5.110? you can also use the portable package, no need to uninstall/reinstall

mattrosenthaltx, 2020-11-08:
I have been running as Administrator. 
I used the portable version and it finally worked after a few tries. Perhaps its a problem with the installable version?

binghaow, 2020-12-24:
@tomgalletti  I have the similar issue, the way I fixed it is by adding some system environmental variables:
1, Type in "Edit System environmental variables" in Windows search
2, System Properties-->Advanced-->Environment Variables
3, Under System variables, find "Path", Click Edit
4, Click "New", add following path
   C:\Windows\System32
   C:\Windows\System32\WindowsPowerShell\v1.0