balena-io/etcher #3347: Update for version 1.5.45 is not available (latest version: 1.5.110, downgrade is disallowed).

rolandweber, 2020-11-08:
I'm running the AppImage of Etcher 1.5.45 on Linux and get this message:

```txt
Update for version 1.5.45 is not available (latest version: 1.5.110, downgrade is disallowed).
```

Looks like the version comparison is broken in 1.5.45.
I suggest to update the downloads on https://www.balena.io/etcher/ to version 1.5.110.

thundron, 2020-11-08:
@rolandweber you likely have javascript disabled, either in general or just for that page - etcher v1.5.45 is the fallback version we use when we can't fetch the latest from github releases (which would otherwise be shown instead).

closed 2020-11-08 by thundron

rolandweber, 2020-11-11:
Correct, I'm surfing with a JavaScript blocker. I had version 1.5.45 still on my machine, from a download earlier this year.

For the records, version 1.5.45 worked just fine for me, where unetbootin and `dd` had failed :-)