balena-io/etcher #3353: Raspberry pi CM4 not recognized.

admindotnu, 2020-11-19:
- **1.5.109** 
- **win10 x64 intel** 
- **none** 
- **Do you see any meaningful error information in the DevTools?** 

Is there a possibility to add support for the cm4 module. We really love this flasher with built-in support for the cm3. Would be nice if the bootcode for cm4 could be added too.

shawaj, 2020-12-08:
@admindotnu this is being worked on. We are just waiting for some of our team to get CM4s and IO boards - the stock availability is not very good at the moment!

admindotnu, 2020-12-11:
Awesome ! Thank you for your reply.

airdrummingfool, 2020-12-11:
@admindotnu would you consider re-opening this issue so there's an easy way to track progress on CM4 support?

shawaj, 2020-12-14:
FYI this is also being worked on here https://github.com/balena-io-modules/node-raspberrypi-usbboot/pull/35

@airdrummingfool do you have a CM4? Might be helpful to have some help testing.

Will re-open for now until we have this support.

airdrummingfool, 2020-12-15:
@shawaj I'm trying hard to get one, but no dice yet. If I get one before this is closed out, I'd be happy to help test.

shawaj, 2020-12-15:
@airdrummingfool yeah we are facing the same issues here! They are a bit hard to come by at the moment.

aessig, 2021-01-07:
@shawaj @airdrummingfool What is the status on this ? We would also love to use etcher. We have CM4 and carrier board for testing if needed. [TOFU Carrier Board](https://twitter.com/oratek_ch/status/1337500145966051330)

shawaj, 2021-01-07:
@aessig that would be a question for @zvin or @thundron to be honest. It is in the roadmap but I'm not sure if we have a timeline yet to be honest.

aessig, 2021-02-11:
@shawaj thanks for your reply. I understand. Please let me know @zvin and @thundron if I can be of any help, to accelerate the process.

airdrummingfool, 2021-03-17:
@shawaj I now have a CM4 and IO Board and would be happy to do testing, or anything else that would help move this forward.

th0f, 2021-04-06:
I think this comes down to implementing the latest https://github.com/balena-io-modules/node-raspberrypi-usbboot. Would love to see this! Saves time in manually running rpiboot.

aessig, 2021-08-29:
Any news? @th0f

tomfrissen, 2021-08-30:
I think this issue can be closed, the latest version has support for this.

aessig, 2021-08-30:
@tomfrissen You are right (needed an upgrade) it works like a charm!

bhjelstrom, 2022-07-30:
HW revision c03141 doesn't seem to be supported (tested on 7 new cm4 units). I think another usbboot update is necessary.