balena-io/etcher #3354: Etched USB stick cannot boot from 32 bit EFI 64 bit CPU machine

thaivic, 2020-11-22:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

thaivic, 2020-11-22:
I used Balena Etcher V1.5.57 (also tried 1.5.109) to produce a 16 G bootable USB stick with linux-lite-5.0-64bit.iso (also tried linux-lite-5.2-64bit.iso, and linux-lite-3.8-32bit.iso). None of them able to boot on WinBook TW700 Windows 8.1 tablet. However, I produced one such USB stick back in June 2020 with the same linux-lite-5.0-64bit.iso and it was able to boot up on this same TW700 tablet but I don't remember which etcher / flasher I used. I no longer have this usable USB stick and I have been unable to reproduce a working one.

lurch, 2020-11-22:
https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable and https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#how-can-i-configure-persistent-storage suggest some alternative image-flashing programs, which may do what you need?

thaivic, 2020-11-22:
Thanks Andrew, I am going to try them out.

Victor

thaivic, 2020-11-23:
Amazing! Rufus worked on the first try. I wish I had tried this before instead of spending 1 week of times messing with it.

closed 2020-11-23 by thundron