balena-io/etcher #3355: Can not write image, SD-card unusable after

efluon, 2020-11-22:
-  Version 1.5.105
- Win 10 / x64 
- 2020-08-20-octopi-buster-arm64-lite-0.18.0.img
-  not really {"stack":"Error: The writer process ended unexpectedly\n    at createError (file:///C:/Users/eike/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:62995:23)\n    at createUserError (file:///C:/Users/eike/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:63006:16)\n    at Server.<anonymous> (file:///C:/Users/eike/AppData/Local/Programs/balena-etcher/resources/app/generated/gui.js:78921:28)","message":"The writer process ended unexpectedly","description":"Please try again, and contact the Etcher team if the problem persists","report":false,"image":"2020-08-20-octopi-buster-arm64-lite-0.18.0.img"}
SD-Card is 8GB, not usable anymore.

Did the same yesterday using the 32bit image from here https://github.com/guysoft/OctoPi/issues/692 . Same error. SD-card was not usable after. Tried formatting in Windows, and with SD card formatter. 16GB card now shows 2.2 TB, but can not be formatted.

Same card was writable with OctoPi image 0.17 before (but that did not run on RPi 4).

Any help? I'm not risking any more cards now..

lurch, 2020-11-22:
>  16GB card now shows 2.2 TB, but can not be formatted.

That sounds very odd - is it possible it was a [fake](https://sosfakeflash.wordpress.com/) SD card?

efluon, 2020-11-22:
> > 16GB card now shows 2.2 TB, but can not be formatted.
> 
> That sounds very odd - is it possible it was a [fake](https://sosfakeflash.wordpress.com/) SD card?
It IS odd. 
1st card was a genuine class 10 SanDisk AFAIK (bought from a respected dealer in Germany). Also, it worked with the other image (was able to edit wpa_supplican and start system on a Pi Zero W). 
Second card is an Emtec/BASF branded 8GB class 10. This is also not accessible any more, directly after a failed image-write.

The difference with the first card is 
a) i did not try the 0.17 image on the second card. 
b) the second card fails any formatting. first card failed on formatting, but has the strange size since then.

thundron, 2020-11-23:
I suggest a read of [this article](https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/) for a rough explanation of what etcher does + solutions for the various OSs to restore the drive so that your OS can read it.
Keep in mind that when you flash an image to a target, the target's content will then be replaced by the content of the image and the OS might not be able to read the new partitions or it might read only part of them, which some refer to as "unusable".

closed 2020-11-23 by thundron

efluon, 2020-11-23:
well, so long, and thanks for the fish.

efluon, 2021-05-19:
Thx, Rufus or classic dd work well for the imaging also. Anything, really, as long as it doesn't use Electron as a smokescreen..