balena-io/etcher #3356: Newest version of etcher broke my SD card

Coaggly, 2020-11-22:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?** 

SD card was working before perfectly now it wont work at all!! Sorry about my tone, i'm on the verge of a break down

shawaj, 2020-11-23:
Which version of etcher are you using? And what were you trying to do with the SD card - flash an image from a file or copy from another device?

Also - Etcher can't break SD cards - it is not possible. 

See here - https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/ and here - https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

Coaggly, 2020-11-24:
1.5.109 (1.5.109), flashing an image

i see this happened to many other ppl too, i'll try the link method

shawaj, 2020-11-24:
@Coaggly the Etcher software just does a block by block copy of the data, so it isn't really possible for it to break your card, as discussed in more detail in the first link I posted.

By the way - there is an [Etcher v1.5.111](https://www.balena.io/etcher/) out now.

Coaggly, 2020-11-26:
> Etcher 1.5.111

I have auto update turned on (slider on right side) and it's not auto updating.

lurch, 2020-11-26:
> i see this happened to many other ppl too

See also https://github.com/pbatard/rufus/wiki/FAQ#help-rufus-damaged-my-flash-drive

shawaj, 2020-11-26:
@Coaggly not every update triggers an auto update https://forums.balena.io/t/how-does-the-auto-update-work/7346

Coaggly, 2020-11-28:
> @Coaggly not every update triggers an auto update https://forums.balena.io/t/how-does-the-auto-update-work/7346

thank you!

shawaj, 2020-11-28:
@Coaggly did you manage to recover the SD card?

Coaggly, 2020-11-30:
> @Coaggly did you manage to recover the SD card?

embarrasingly, while i thought it was broken, i went through my storage unit and got a new SD card and left the old one there. Next time i go past i'll have to remember to try the fix.

shawaj, 2020-12-08:
@ghost ok let us know how you get on

shawaj, 2020-12-08:
Closing as user has deleted their account

closed 2020-12-08 by shawaj