balena-io/etcher #3357: Tails Balena etcher cant select flash drive

BidenLovesMinors, 2020-11-23:
- **Etcher version:1.5.105 macOS

- **Image flashed: tails-amd64-4.13.img 

Whenever i go and try and select the drive it wont let me even select it i dont know if im just dumb or what.

[image]

 `Cmd+Opt+I` if you're on macOS. -->

lurch, 2020-11-23:
I'm not an Etcher developer, but I think the "Source drive" label might mean that the `tails-amd64-4.13.img` image you're trying to flash is stored on your "Lexar USB Flash Drive Media"? You can't read from the drive at the same time you're overwriting it :wink:  If my guess is correct, the problem should go away if you copy the image file to your HardDrive and select it from there instead?

thundron, 2020-11-23:
That means you're trying to select a target that contains the source image that you selected.
As @lurch mentioned, try copying the image on a different drive so you can select the target you need to flash.
Just keep in mind that the content of the target will be entirely replaced by the content of the image

closed 2020-11-23 by thundron

sajolida, 2021-12-22:
Let's see if we can fix the root cause of bad feedback on #3662.