balena-io/etcher #3358: Won't Process ISO File

Bl2c5FlA9, 2020-11-23:
- **Etcher version:** 1.5.109 (1.5.109)
- **Operating system and architecture:** macOS 10.12.6
- **Image flashed:** <none>
- **Do you see any meaningful error information in the DevTools?** 

No matter what ISO file I present to etcher it tells me the ISO is either incomplete or corrupt.  That is not the case.  I have used the exact same ISO multiple times to make bootable images.  So either you have a stop missing in your usage procedure or you just report that every ISO is corrupt.  I have used multiple ISO files with the same result.  I see no evidence that this product works at all.

shawaj, 2020-11-23:
@Bl2c5FlA9 are any of the ISO files you have tried available to download publicly? So we can try to reproduce the issue.

31piy, 2020-11-24:
I've tried this with the latest Ubuntu 20.04 ISO available on their site, and it gives me the same error with the latest version of Etcher. The same image works with v1.5.109 though.

shawaj, 2020-11-24:
@31piy what machine are you using? And the operating system version?

And the image is 20.04.1 LTS? From here - https://ubuntu.com/download/desktop

You are having an issue with Etcher v1.5.111 but it works in v1.5.109?

Do you get any error message?

myrrh0, 2020-11-24:
i'm having the same issue. using macOS Catalina 10.15.4. the ISO i'm trying to use is manjaro 10.1 GNOME (image here https://manjaro.org/downloads/official/gnome/). 
using etcher v1.5.111

31piy, 2020-11-25:
@shawaj I'm using MacOS Catalina v10.15.7. Was trying to flash [this image](https://releases.ubuntu.com/20.04/ubuntu-20.04.1-desktop-amd64.iso) using Etcher v1.5.111. It failed stating the reason that the image is corrupted.

I downgraded Etcher to v1.5.109 and the same image could be flashed without any issues.

lxne, 2020-12-01:
Same issue here with Catalina 10.15.7 and Etcher (1.5.109 and 1.5.111). Tried three different iso files (hefftorlinux, linuxmint, elementary). Tried flashing an USB stick and a sdcard. I always get the error message as in the screenshot.
[image]

shawaj, 2020-12-01:
@lxne @31piy @Bl2c5FlA9 I managed to reproduce this issue once on macOS Big Sur 11.0.1 (20B29) - however it only fails in this way when downloading the .iso and trying to image using that using Etcher - which has been open for some time. However, I force-quit etcher and it then succeeds.

If I use the `Flash from URL` option and use the download URL - it seems to work fine every time.

Could you try these suggestions and see if it fixes anything?

lurch, 2020-12-01:
@shawaj I don't have a Mac, but...

> however it only fails in this way when downloading the .iso and trying to image using that using Etcher - which has been open for some time. However, I force-quit etcher and it then succeeds.

...makes it sound like if you start Etcher first, and then download an ISO (with another application), the still-running Etcher isn't able to read the freshly-downloaded ISO; but if the ISO is downloaded _before_ you start Etcher, then Etcher **is** able to read the ISO? Almost sounds like the OS is incorrectly "caching" the file-permissions on the ISO file? :man_shrugging:

shawaj, 2020-12-01:
@lurch let me try that now...

lxne, 2020-12-01:
> If I use the `Flash from URL` option and use the download URL - it seems to work fine every time.

This works for me too.

But force quitting and trying the downloaded iso again is not working.

shawaj, 2020-12-01:
@lxne @lurch weirdly, I can't reproduce the .iso error at all now on v1.5.111

lurch, 2020-12-01:
@shawaj Have you tried both an admin and a non-admin user account?

@lxne In various other Etcher-on-Catalina issues, some people say that running Etcher with `sudo` works, and some people say that copying the ISO directly into Etcher's application-folder works :shrug: :green_apple:

lxne, 2020-12-01:
> @lxne In various other Etcher-on-Catalina issues, some people say that running Etcher with `sudo` works, and some people say that copying the ISO directly into Etcher's application-folder works 🤷 🍏

I tried a view things with the iso files. First I moved all of them into the users/shared folder, but that didn't help. Then I copied one of the iso files on a sdcard and that did the trick. Flashing it from the sdcard to the usb stick worked.

shawaj, 2020-12-01:
@lxne so that was just with a raw iso stored on an sd card? rather than an imaged SD card?

lurch, 2020-12-01:
@shawaj From what I've been able to surmise from various Etcher issues, it seems MacOS is very pernickety about file-permissions / access-rights / sandboxing :confused:  It probably worked (most recently) for @lxne because the SDCard was just in FAT32-format, which doesn't support the same file-permissions as MacOS' native filesystems.

lxne, 2020-12-01:
> @lxne so that was just with a raw iso stored on an sd card? rather than an imaged SD card?

Yes, just a raw iso file copied onto a sdcard (which is formatted as exfat).
I also think it is a file permission issue.

shawaj, 2020-12-01:
@lurch ah that gives me something else to test. I didn't wipe the SD card between writes so maybe that is our

shawaj, 2020-12-02:
@lurch bingo - it fails in the same way if it is initialised as a clean SD card (using SD card association formatter) as you suggested.

[image]
[image]
[image]

lurch, 2020-12-02:
@shawaj There's another MacOS-related Etcher issue I saw recently where people were saying that they had to unmount the drive, before Etcher would succeed... https://github.com/balena-io/etcher/issues/2933#issuecomment-736861573

It sounds like there's _probably_ multiple different MacOS issues affecting different people, as some people claim one thing fixes it for them, but other people claim only a different thing fixes it for them? :shrug:

shawaj, 2020-12-02:
@lurch yes if I unmount from disk utility then it works perfectly. I guess this is the same issue as having it clean/formatted - as the bootable ubuntu SD doesn't auto-mount on mac so it has the same effect as unmounting. 

The weird thing is that this only seems to happen with when flashing from file - not when doing it from URL. Just for completeness - it is not just ISO files either - it happens with .zip and .img as well from my tests.

I think the force quit thing I mentioned above may have been a red-herring as I can't reproduce it now.

shawaj, 2020-12-02:
Edit: updated the title to more accurately reflect the issue

shawaj, 2020-12-03:
@lxne @31piy @Bl2c5FlA9 this version is working for me - https://github.com/balena-io/etcher/releases/download/untagged-1e30b3a0b27ec01ea938/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-03:
@lxne @31piy @Bl2c5FlA9 here is the correct link - https://github.com/balena-io/etcher/releases/download/v1.5.112/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-08:
@lxne @31piy @Bl2c5FlA9 is this working now for you in v1.5.112?

lxne, 2020-12-08:
> @lxne @31piy @Bl2c5FlA9 is this working now for you in v1.5.112?

Sorry, didn't have the time to test it earlier.
My first test was successful. I think it is working correctly now.

Thank you!

shawaj, 2020-12-08:
Great! Will close this issue now then.

Feel free to reply if you disagree.

closed 2020-12-08 by shawaj