balena-io/etcher #3363: MacOS: Big Sur - The Writer Process Ended Unexptec

RedHoodedWraith, 2020-11-25:
Hi,

After updating to version 1.5.111, I have found that I am unable to flash anything onto a MicroSD card. I have formatted the card to FAT format. Below is some information about errors that occur everytime I try to flash a card. A few minutes prior to updating to version 1.5.111, I was using version 1.5.110 on the same operating system and I was able to successfully flash a `.img.xz` file onto a microSD card. I suspect a minor change in the latest version (1.5.11) may be causing this issue.

- **Etcher version:** 1.5.111
- **Operating system and architecture:** MacOS Big Sur 11.0.1
- **Image flashed:** `.img.xz` and `.iso`
- **Do you see any meaningful error information in the DevTools?** 
When attempting to flash a `.img.xz` file, the following occurred:

```
{"stack":"Error: Can't configure or trim a source that is not randomly readable, skipping\nError: EBADF: bad file descriptor, read\n/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:4074\n                throw ex;\n                ^\n\nRuntimeError: abort(Error: EBADF: bad file descriptor, read). Build with -s ASSERTIONS=1 for more info.\n    at process.abort (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:4453:17)\n    at process.emit (events.js:228:7)\n    at processPromiseRejections (internal/process/promises.js:201:33)\n    at processTicksAndRejections (internal/process/task_queues.js:95:32)\n\n    at createError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:71076:23)\n    at elevateScriptCatalina (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:87259:19)\n    at async withTmpFile (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:85838:20)\n    at async elevateCommand (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:87275:16)\n    at async Server.<anonymous> (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:87428:37)","message":"Can't configure or trim a source that is not randomly readable, skipping\nError: EBADF: bad file descriptor, read\n/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:4074\n                throw ex;\n                ^\n\nRuntimeError: abort(Error: EBADF: bad file descriptor, read). Build with -s ASSERTIONS=1 for more info.\n    at process.abort (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:4453:17)\n    at process.emit (events.js:228:7)\n    at processPromiseRejections (internal/process/promises.js:201:33)\n    at processTicksAndRejections (internal/process/task_queues.js:95:32)\n","image":"ubuntu-20.04.1-5.4-mate-odroid-xu4-20200818.img.xz"}
```

When attempting to flash a `.iso` file, the following occured:
```
{"stack":"Error: The writer process ended unexpectedly\n    at createError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:71076:23)\n    at createUserError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:71087:16)\n    at Server.<anonymous> (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:87446:28)","message":"The writer process ended unexpectedly","description":"Please try again, and contact the Etcher team if the problem persists","report":false,"image":"ubuntu-20.04.1-live-server-amd64.iso"}
```

flochtililoch, 2020-11-26:
Also getting this on High Sierra, with a `.img` file with 1.5.111. Same card / same image on 1.5.110 works fine.

Krisalexroberts, 2020-11-26:
Same issue, however retrying works fine.

Consultant99, 2020-12-02:
Same for Mojave 10.14.5 on a new flash card, (which worked fine the 1st time, but not the second)

|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.111, Type = dmg
gui.js:64192 Tue Dec 01 2020 20:18:33 GMT-0500 (Eastern Standard Time) Application start ({"packageType":"dmg","version":"1.5.111","sample":0.1,"applicationSessionUuid":"aad8832d-8a94-4171-aebd-26d0b2a2971d","flashingWorkflowUuid":"4a54b72b-daad-4569-81f1-f140280449fc"})
gui.js:56894 Tue Dec 01 2020 20:19:30 GMT-0500 (Eastern Standard Time) Open image selector ({"sample":0.1,"applicationSessionUuid":"aad8832d-8a94-4171-aebd-26d0b2a2971d","flashingWorkflowUuid":"4a54b72b-daad-4569-81f1-f140280449fc"})
gui.js:56894 Tue Dec 01 2020 20:19:37 GMT-0500 (Eastern Standard Time) Select image ({"image":{"size":1064337408,"name":"DietPi_RPi-ARMv6-Buster.img","hasMBR":true,"partitions":[{"offset":4194304,"size":268435456,"type":12,"index":1},{"offset":272629760,"size":791576576,"type":131,"index":2}],"extension":"img","path":"/Users/leslidtke/Downloads/DietPi_RPi-ARMv6-Buster/DietPi_RPi-ARMv6-Buster.img","logo":false,"blockMap":false},"sample":0.1,"applicationSessionUuid":"aad8832d-8a94-4171-aebd-26d0b2a2971d","flashingWorkflowUuid":"4a54b72b-daad-4569-81f1-f140280449fc"})
gui.js:56894 Tue Dec 01 2020 20:22:11 GMT-0500 (Eastern Standard Time) Toggle drive ({"drive":{"size":31914983424,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/XHC1@14","mountpoints":[{"path":"/Volumes/UNTITLED","label":"UNTITLED"}],"description":"Generic USB SD Reader Media","isUAS":null,"partitionTableType":"mbr"},"previouslySelected":false,"sample":0.1,"applicationSessionUuid":"aad8832d-8a94-4171-aebd-26d0b2a2971d","flashingWorkflowUuid":"4a54b72b-daad-4569-81f1-f140280449fc"})
gui.js:56894 Tue Dec 01 2020 20:22:15 GMT-0500 (Eastern Standard Time) Flash ({"image":{"size":1064337408,"name":"DietPi_RPi-ARMv6-Buster.img","hasMBR":true,"partitions":[{"offset":4194304,"size":268435456,"type":12,"index":1},{"offset":272629760,"size":791576576,"type":131,"index":2}],"extension":"img","path":"/Users/leslidtke/Downloads/DietPi_RPi-ARMv6-Buster/DietPi_RPi-ARMv6-Buster.img"},"drives":[{"size":31914983424,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/XHC1@14","mountpoints":[{"path":"/Volumes/UNTITLED","label":"UNTITLED"}],"description":"Generic USB SD Reader Media","isUAS":null,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"a1ac8909-15b2-436c-9ba0-9f474848984e","status":"started","flashInstanceUuid":"a1ac8909-15b2-436c-9ba0-9f474848984e","unmountOnSuccess":true,"sample":0.1,"applicationSessionUuid":"aad8832d-8a94-4171-aebd-26d0b2a2971d","flashingWorkflowUuid":"4a54b72b-daad-4569-81f1-f140280449fc"})
gui.js:56894 Elevating command: /Applications/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper (Renderer).app/Contents/MacOS/balenaEtcher Helper (Renderer) /Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js
gui.js:56894 Tue Dec 01 2020 20:22:15 GMT-0500 (Eastern Standard Time) SafeWebview loaded ({"event":{"id":1,"url":"https://assets.balena.io/etcher-featured/index.html?borderRight=false&darkBackground=true&etcher-version=1.5.111&api-version=2&optOutAnalytics=false","method":"GET","timestamp":1606872135434.5852,"resourceType":"mainFrame","ip":"104.31.83.157","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"access-control-allow-origin":["*"],"age":["417"],"cache-control":["max-age=600"],"cf-cache-status":["DYNAMIC"],"cf-ray":["5fb13c5e2f71e730-EWR"],"cf-request-id":["06c2a40ed90000e7309fae5000000001"],"content-encoding":["br"],"content-type":["text/html; charset=utf-8"],"date":["Wed, 02 Dec 2020 01:22:15 GMT"],"expect-ct":["max-age=604800, report-uri=\"https://report-uri.cloudflare.com/cdn-cgi/beacon/expect-ct\""],"expires":["Fri, 27 Nov 2020 08:37:04 GMT"],"last-modified":["Thu, 26 Nov 2020 10:50:28 GMT"],"nel":["{\"report_to\":\"cf-nel\",\"max_age\":604800}"],"report-to":["{\"endpoints\":[{\"url\":\"https:\\/\\/a.nel.cloudflare.com\\/report?s=XOmZO593Jj%2FzZ%2FVpiJN1KSTAzvdePGbGLgyfnjXJMx0pgn9CFgIHrIPfiD7nx9FeYRzAL3mKxc7YBYlay%2FNK%2BnNyMlnAWWrknGCUW8joNajU\"}],\"group\":\"cf-nel\",\"max_age\":604800}"],"server":["cloudflare"],"status":["200"],"vary":["Accept-Encoding"],"via":["1.1 varnish"],"x-cache":["HIT"],"x-cache-hits":["1"],"x-fastly-request-id":["704dbcebf6d1910719aa2abd23698f9726552577"],"x-github-request-id":["9D84:3BC8:1F24AF:26930C:5FC0B918"],"x-proxy-cache":["HIT"],"x-served-by":["cache-ewr18125-EWR"],"x-timer":["S1606872135.401892,VS0,VE1"]},"referrer":"","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"aad8832d-8a94-4171-aebd-26d0b2a2971d","flashingWorkflowUuid":"4a54b72b-daad-4569-81f1-f140280449fc"})
[... cut ...]
async function (async)
flashImageToDrive @ gui.js:87635
tryFlash @ gui.js:87730
callback @ gui.js:87752
ba @ gui.js:50437
ja @ gui.js:50451
ka @ gui.js:50454
oa @ gui.js:50468
lc @ gui.js:51139
jc @ gui.js:51133
mc @ gui.js:51150
sc @ gui.js:51219
Ia @ gui.js:56449
Ma @ gui.js:50547
Rc @ gui.js:51485
id @ gui.js:51454
exports.unstable_runWithPriority @ gui.js:56778
cg @ gui.js:52646
Ga @ gui.js:56437
gd @ gui.js:51444
gui.js:6038  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices

Consultant99, 2020-12-02:
Worked on older etcher version, same sd card

Confirm 1.5.109 works properly, there is an issue in 1.5.111.

Consultant99, 2020-12-02:
Check the version - 1.5.111 has issues with macs,

lurch, 2020-12-02:
I think I remember @zvin writing that he was planning to release 1.5.112 soon?

Consultant99, 2020-12-02:
Andrew
Not sure, I just came into my 1st raspberry software install,
thought I was going nuts with reformatting several SD cards for use with etcher,
spent more time on that than the actual diet-pi install.
I had a etcher release from several days ago, that .109 version,
and it worked fine.
Les

shawaj, 2020-12-03:
@lurch @Consultant99 @RedHoodedWraith - https://github.com/balena-io/etcher/releases/download/untagged-1e30b3a0b27ec01ea938/balenaEtcher-1.5.112.dmg

Also see #3358

Consultant99, 2020-12-03:
Had it happen on the older 109 version too..even after re-installing 109.
Also was trying SD Card formatter besides using MAC OS Disk Util - Etcher still had issues.
Finally worked - I think after both formatting utilities have to closed.

shawaj, 2020-12-03:
@lurch @Consultant99 @RedHoodedWraith here is the correct link - https://github.com/balena-io/etcher/releases/download/v1.5.112/balenaEtcher-1.5.112.dmg

Consultant99, 2020-12-03:
shawl, downloaded .112 , ran it three times fine on the same card (exited and restarted etcher each time, but used same SD)

seems like its working now

lurch, 2020-12-04:
@shawaj As I've mentioned elsewhere, I don't have a Mac myself so the DMG is of little use to me :laughing:

shawaj, 2020-12-08:
@lurch oops! My bad

shawaj, 2020-12-08:
Closing as seems to be fixed in 1.5.112

closed 2020-12-08 by shawaj