balena-io/etcher #3366: Unable to put an ISO image onto a USB key, resource busy or locked

georgiats, 2020-11-26:
The user is unable to put an ISO image onto a USB key. The error suggests the key is busy or locked
It's a transient error, retrying opening the file after a slight delay fixes it. This should be fixed in the next release.

- **Etcher version:**  balenaEtcher version is 1.5.111
- **Operating system and architecture:**  Mac OS X version is 10.14.6.
- **Image flashed:** Clonezilla iso image on a 128GB USB Key.

```gui.js:88077  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.5.111, Type = dmg
gui.js:64192 Tue Nov 24 2020 15:03:33 GMT+0100 (Ora standard dell’Europa centrale) Application start ({"packageType":"dmg","version":"1.5.111","sample":0.1,"applicationSessionUuid":"38163fb3-ff3e-4e03-a1ae-f31ecd490f9b","flashingWorkflowUuid":"245a20de-07fe-4c83-bfe9-b1b59c22ce47"})
gui.js:56894 Tue Nov 24 2020 15:03:47 GMT+0100 (Ora standard dell’Europa centrale) Open image selector ({"sample":0.1,"applicationSessionUuid":"38163fb3-ff3e-4e03-a1ae-f31ecd490f9b","flashingWorkflowUuid":"245a20de-07fe-4c83-bfe9-b1b59c22ce47"})
gui.js:56894 Tue Nov 24 2020 15:04:26 GMT+0100 (Ora standard dell’Europa centrale) Select image ({"image":{"size":299892736,"name":"clonezilla-live-2.7.0-10-i686.iso","hasMBR":true,"partitions":[{"offset":32768,"size":299859968,"type":23,"index":1}],"extension":"iso","path":"/Users/marco/Downloads/clonezilla-live-2.7.0-10-i686.iso","logo":false,"blockMap":false},"sample":0.1,"applicationSessionUuid":"38163fb3-ff3e-4e03-a1ae-f31ecd490f9b","flashingWorkflowUuid":"245a20de-07fe-4c83-bfe9-b1b59c22ce47"})
gui.js:56894 Tue Nov 24 2020 15:04:30 GMT+0100 (Ora standard dell’Europa centrale) Toggle drive ({"drive":{"size":128043712512,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/","mountpoints":[{"path":"/Volumes/LOFFIO","label":"LOFFIO"}],"description":"SanDisk Extreme Pro Media","isUAS":null,"partitionTableType":"mbr"},"previouslySelected":false,"sample":0.1,"applicationSessionUuid":"38163fb3-ff3e-4e03-a1ae-f31ecd490f9b","flashingWorkflowUuid":"245a20de-07fe-4c83-bfe9-b1b59c22ce47"})
gui.js:56894 Tue Nov 24 2020 15:04:35 GMT+0100 (Ora standard dell’Europa centrale) Flash ({"image":{"size":299892736,"name":"clonezilla-live-2.7.0-10-i686.iso","hasMBR":true,"partitions":[{"offset":32768,"size":299859968,"type":23,"index":1}],"extension":"iso","path":"/Users/marco/Downloads/clonezilla-live-2.7.0-10-i686.iso"},"drives":[{"size":128043712512,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/","mountpoints":[{"path":"/Volumes/LOFFIO","label":"LOFFIO"}],"description":"SanDisk Extreme Pro Media","isUAS":null,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"b5a239e2-5c35-4054-8d5a-0c09d000a593","status":"started","flashInstanceUuid":"b5a239e2-5c35-4054-8d5a-0c09d000a593","unmountOnSuccess":true,"sample":0.1,"applicationSessionUuid":"38163fb3-ff3e-4e03-a1ae-f31ecd490f9b","flashingWorkflowUuid":"245a20de-07fe-4c83-bfe9-b1b59c22ce47"})
gui.js:56894 Elevating command: /Applications/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper (Renderer).app/Contents/MacOS/balenaEtcher Helper (Renderer) /Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js
gui.js:56894 Successfully connected to IPC server: etcher-server-20698, socket root /tmp/
gui.js:56894 Image: /Users/marco/Downloads/clonezilla-live-2.7.0-10-i686.iso
gui.js:56894 Devices: /dev/disk2
gui.js:56894 Umount on success: true
gui.js:56894 Auto blockmapping: true
gui.js:56894 Decompress first: true
gui.js:56894 Tue Nov 24 2020 15:04:35 GMT+0100 (Ora standard dell’Europa centrale) Flash error ({"image":{"size":299892736,"name":"clonezilla-live-2.7.0-10-i686.iso","hasMBR":true,"partitions":[{"offset":32768,"size":299859968,"type":23,"index":1}],"extension":"iso","path":"/Users/marco/Downloads/clonezilla-live-2.7.0-10-i686.iso"},"drives":[{"size":128043712512,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/","mountpoints":[{"path":"/Volumes/LOFFIO","label":"LOFFIO"}],"description":"SanDisk Extreme Pro Media","isUAS":null,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"b5a239e2-5c35-4054-8d5a-0c09d000a593","flashInstanceUuid":"b5a239e2-5c35-4054-8d5a-0c09d000a593","unmountOnSuccess":true,"error":{"name":"Error","message":"EBUSY: resource busy or locked, open '/dev/rdisk2'","stack":"Error: EBUSY: resource busy or locked, open '/dev/rdisk2'","code":"EBUSY","syscall":"open","errno":-16},"sample":0.1,"applicationSessionUuid":"38163fb3-ff3e-4e03-a1ae-f31ecd490f9b","flashingWorkflowUuid":"245a20de-07fe-4c83-bfe9-b1b59c22ce47"})
gui.js:56894 Finish: 0
gui.js:6038  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:56894 Terminating IPC server
gui.js:56894 Flash results {results: {…}, cancelled: false, skip: false}
gui.js:56894 Tue Nov 24 2020 15:04:37 GMT+0100 (Ora standard dell’Europa centrale) Write failed ({"image":{"size":299892736,"name":"clonezilla-live-2.7.0-10-i686.iso","hasMBR":true,"partitions":[{"offset":32768,"size":299859968,"type":23,"index":1}],"extension":"iso","path":"/Users/marco/Downloads/clonezilla-live-2.7.0-10-i686.iso"},"drives":[{"size":128043712512,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/","mountpoints":[{"path":"/Volumes/LOFFIO","label":"LOFFIO"}],"description":"SanDisk Extreme Pro Media","isUAS":null,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"b5a239e2-5c35-4054-8d5a-0c09d000a593","status":"failed","flashInstanceUuid":"b5a239e2-5c35-4054-8d5a-0c09d000a593","unmountOnSuccess":true,"error":{"description":"Please try again, and contact the Etcher team if the problem persists","report":false},"sample":0.1,"applicationSessionUuid":"38163fb3-ff3e-4e03-a1ae-f31ecd490f9b","flashingWorkflowUuid":"245a20de-07fe-4c83-bfe9-b1b59c22ce47"})
gui.js:56894 {"stack":"Error: The writer process ended unexpectedly\n    at createError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:71076:23)\n    at createUserError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:71087:16)\n    at Server.<anonymous> (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:87446:28)","message":"The writer process ended unexpectedly","description":"Please try again, and contact the Etcher team if the problem persists","report":false,"image":"clonezilla-live-2.7.0-10-i686.iso"}
console.<computed> @ gui.js:56894
logException @ gui.js:64262
flashImageToDrive @ gui.js:87651
async function (async)
flashImageToDrive @ gui.js:87635
handleWarningResponse @ gui.js:87691
done @ gui.js:87770
ba @ gui.js:50437
ja @ gui.js:50451
ka @ gui.js:50454
oa @ gui.js:50468
lc @ gui.js:51139
jc @ gui.js:51133
mc @ gui.js:51150
sc @ gui.js:51219
Ia @ gui.js:56449
Ma @ gui.js:50547
Rc @ gui.js:51485
id @ gui.js:51454
exports.unstable_runWithPriority @ gui.js:56778
cg @ gui.js:52646
Ga @ gui.js:56437
gd @ gui.js:51444```

shawaj, 2020-12-03:
Try this - https://github.com/balena-io/etcher/releases/download/untagged-1e30b3a0b27ec01ea938/balenaEtcher-1.5.112.dmg

Also relates to #3358 I believe

shawaj, 2020-12-03:
Here is the correct link - https://github.com/balena-io/etcher/releases/download/v1.5.112/balenaEtcher-1.5.112.dmg

zvin, 2020-12-07:
This should be fixed in 1.5.112

closed 2020-12-07 by zvin

Joshuah143, 2021-08-20:
Just experienced the same issue on 1.5.112