balena-io/etcher #3367: Unable to flash Ubuntu 20.1 amd64 to SD card

cedricve, 2020-11-27:
- **Etcher version:**  1.5.11
- **Operating system and architecture:**  Mac OSX
- **Image flashed:** https://ubuntu.com/download/desktop
- **Do you see any meaningful error information in the DevTools?** 

Shortly after pressing the flash button the error popups. My purpose is to flash ubuntu to a SD card, and use it as a bootable device for an Intel NUC.

gui.js:56894 {"stack":"Error: Can't configure or trim a source that is not randomly readable, skipping\nError: EBADF: bad file descriptor, read\n/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:4074\n                throw ex;\n                ^\n\nRuntimeError: abort(Error: EBADF: bad file descriptor, read). Build with -s ASSERTIONS=1 for more info.\n    at process.abort (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:4453:17)\n    at process.emit (events.js:228:7)\n    at processPromiseRejections (internal/process/promises.js:201:33)\n    at processTicksAndRejections (internal/process/task_queues.js:95:32)\n\n    at createError (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:71076:23)\n    at elevateScriptCatalina (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:87259:19)\n    at async withTmpFile (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:85838:20)\n    at async elevateCommand (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:87275:16)\n    at async Server.<anonymous> (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:87428:37)","message":"Can't configure or trim a source that is not randomly readable, skipping\nError: EBADF: bad file descriptor, read\n/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:4074\n                throw ex;\n                ^\n\nRuntimeError: abort(Error: EBADF: bad file descriptor, read). Build with -s ASSERTIONS=1 for more info.\n    at process.abort (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:4453:17)\n    at process.emit (events.js:228:7)\n    at processPromiseRejections (internal/process/promises.js:201:33)\n    at processTicksAndRejections (internal/process/task_queues.js:95:32)\n","image":"balenaEtcher-1.5.111.dmg"}
console.<computed> @ gui.js:56894
gui.js:6038  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices

lerigner, 2020-11-29:
Etcher version: 1.5.**111**
macOS: 10.14.6
I guess you meant version 1.5.111.
I have a similar issue when trying to flash Ubuntu 20.04.1 LTS server or desktop, Ubuntu 20.10 server or desktop to a USB drive.

Downgrading to Etcher version 1.5.**110** works properly

cedricve, 2020-11-29:
@lerigner thanks for the help will try out, what error do you experience?

cedricve, 2020-11-29:
Confirm 1.5.110 works properly, there is an issue in 1.5.111.

shawaj, 2020-12-03:
@lerigner @cedricve I believe this is the same issue as #3358 

Also - try this version - https://github.com/balena-io/etcher/releases/download/untagged-1e30b3a0b27ec01ea938/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-03:
@lerigner @cedricve here is the correct link - https://github.com/balena-io/etcher/releases/download/v1.5.112/balenaEtcher-1.5.112.dmg

shawaj, 2020-12-08:
@lerigner @cedricve is this working now for you?

shawaj, 2020-12-15:
Will close this as no reply in a while. Feel free to reply back if you don't agree or if you have further issues.

closed 2020-12-15 by shawaj