balena-io/etcher #3369: Privilege Elevation Fails with Non-Admin Account on MacOS

pahowes, 2020-11-27:
Very simply, I log into my Mac with an unprivileged account. it does not have sudo access or administrative rights at all. the new balenaEtcher UI only asks for a password, but not a username. This is in stark contrast to every other macOS application which prompts for both when elevating. The only way I can use this application is by switching over to another user that does have admin rights. This is an unacceptable oversight.

[image]

lurch, 2020-11-27:
I believe it's a "known problem" with Catalina, see e.g. https://github.com/balena-io/etcher/issues/3107#issuecomment-602047807

thundron, 2020-12-03:
We're aware of the discrepancy, but are currently unable to work on it. For more info please look at https://github.com/balena-io/etcher/issues/3321#issuecomment-707352408

As stated there, PRs are welcome!

closed 2020-12-03 by thundron