balena-io/etcher #3372: Attempting to clone doesn't work because target drive is too small, however the difference is negligible

aidenmitchell, 2020-12-02:
- **Etcher version:** 1.5.111
- **Operating system and architecture:** Windows 10, 64-bit
- **Image flashed:** N/A, cloning an image
- **Do you see any meaningful error information in the DevTools?** N/A

I have 2 SD cards, original is 32GB, target is 31.9 GB. Etcher won't clone it because the target is too small, which is true, but there is definitely empty space on the original. Is there a was to bypass that check? Thanks!

lurch, 2020-12-03:
Even if there's "empty space" in the source image, I believe there's no guarantee that the empty space is at the very end of the drive rather than in the middle of the drive? i.e. there may _actually_ be important data in that last 0.1GB ? :shrug:

shawaj, 2020-12-03:
@aidenmitchell could you remove the empty space / shrink the source drive?

lurch, 2020-12-03:
> shrink the source drive?

https://www.youtube.com/watch?v=je0uLWfKh4U :wink:

shawaj, 2020-12-03:
@lurch exactly :-)

thundron, 2020-12-03:
It's still more data than what Etcher would be allowed to write, i.e. you can't fit 4cm in 3.9cm, even if the difference is negligible, given there's no way Etcher can possibly know what data to skip (outside of some exceptions, like ISOs).
You can either increase the target data size or, as @shawaj mentioned, reduce the source data size.

closed 2020-12-03 by thundron