balena-io/etcher #3373: After successful flash, the usb turn into read only file system, this not make sense

biaocy, 2020-12-03:
- **Etcher version:** 1.5.111-x64
- **Operating system and architecture:** MXLinux, `Linux billpc 4.19.0-12-amd64 #1 SMP Debian 4.19.152-1 (2020-10-18) x86_64 GNU/Linux
`
- **Image flashed:** xubuntu-20.04.1-desktop-amd64.iso

After successful flash, the usb turn into **read only** file system, and this not make sense. Tools like **rufus** didn't make the usb read only after successful flash. I'm mean there still a lot of space to put other file into it.

If this is what this tool's intend, you need to tell the user before they flash.

lurch, 2020-12-03:
See #413 
(and also https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md and https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/ )

biaocy, 2020-12-03:
So, etcher is not a good choice, as for now

closed 2020-12-03 by biaocy