balena-io/etcher #3374: Etcher AppImage registers itself as default text/html mime handler

dlech, 2020-12-04:
- **Etcher version:** balenaEtcher-1.5.112-x64.AppImage
- **Operating system and architecture:** Ubuntu 20.04 amd64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

I use the `xdg-open` command line tool to open `.html` files. However each time I run the Etcher AppImage, it registers itself as the default handler for the `text/html` mime type so running `xdg-open my-file.html` launches Etcher with an error code.

Steps to reproduce:
1. Verify current mime handler: `xdg-mime query default text/html`
2. If it is already `balena-etcher-electron.desktop`, replace it with `xdg-mime default firefox.desktop text/html` then repeat step 1 to confirm that it is now `firefox.desktop` instead of ``balena-etcher-electron.desktop`.
3. Launch Etcher
4. Repeat step 1 and see that it has been changed to `balena-etcher-electron.desktop`

lurch, 2020-12-05:
How weird, I wonder if that might be an Electron bug or an AppImage bug? :shrug: 

Just did a quick search and found https://github.com/electron/electron/issues/20382 which sounds very similar.

dlech, 2020-12-05:
Your search skills must be better than mine, nice find :smile:

dlech, 2020-12-06:
Looks like this got reported upstream a few days ago: https://gitlab.freedesktop.org/xdg/xdg-utils/-/issues/180

As a workaround, I commented out this line: https://gitlab.freedesktop.org/xdg/xdg-utils/-/blob/v1.1.3/scripts/xdg-settings.in#L691

    sudo vi $(which xdg-settings)

aldomann, 2021-01-03:
@dlech's do you need to reboot the system for it to work? Just commenting the line didn't fix it for me.

dlech, 2021-01-03:
I don't think a reboot is required. Did you also run `xdg-mime default firefox.desktop text/html` to repair the damage that is already done? Editing `xdg-settings` just prevents it from happening again.

aldomann, 2021-01-03:
My bad. Running `xdg-mime default org.chromium.Chromium.desktop text/html` fixed it for me.

tzarebczan, 2021-01-26:
Anyone know why this happens though? From the electron issue, it sounded like it was related to something with a - in the file name / package, but we're having a similar at LBRY and can't find out the root cause either.

dlech, 2021-01-26:
> Anyone know why this happens though?

It is a bug in xdg-utils: https://github.com/balena-io/etcher/issues/3374#issuecomment-739558436

closed 2021-05-01 by thundron