balena-io/etcher #3375: Etcher unable to create bootable OS X Yosemite disks

Mennaruuk, 2020-12-05:
- **Etcher version:** 1.5.112 
- **Operating system and architecture:** Windows 10 64-bit
- **Image flashed:** InstallMacOSX.dmg
- **Do you see any meaningful error information in the DevTools?**
[image]

Etcher is unable to create a working bootable flash drive out of the official OS X Yosemite .dmg file. The .dmg file in question is InstallMacOSX.dmg (renamed above to Yosemite.dmg). It can be obtained directly from Apple's website [here](http://updates-http.cdn-apple.com/2019/cert/061-41343-20191023-02465f92-3ab5-4c92-bfe2-b725447a070d/InstallMacOSX.dmg). I verified the checksum using two different software as:

**MD5**: 453536DDF8406CFC40EC776DABF3491E
**SHA-1**: C542681961CC7FAFB9C15BFD4AF152A3A57F07FA
**SHA-256**: DE869907CE4289FE948CBD2DEA7479FF9C369BBF47B06D5CB5290D78FB2932C6
**SHA-512**: 1B217F2D5835FF1C7ABC260A369E2ADD0AA9C7A9BE65FC50DD30D69F22F742C8F469E26932B1378F4CAFBD10D02050DA2227E2202564F9F029CC230E846CCCB8

After plugging the USB drive into my Mac and holding Option key at startup, I am met with a white screen and a black cursor with nothing to choose, indicating that the disk isn't properly bootable.

I would like to note that I also have the .pkg file found inside this .dmg file. I opened a VM image of Mac on my PC to properly extract this .pkg file out of the .dmg file, then I transferred it over from the VM to my PC. These are the checksums:

**MD5**: F8959FA719B9A74C63E972FE1A1441E7
**SHA-1**: 39C52FF92A4C965B4CA4608A1788732165164ABB
**SHA-256**: 7E02F978556ECDD3C91ECF03C779E4A1402E41CA7EE2A268F96B7CFB8DF6CA79
**SHA-512**: 83632764A5023A19F61EB229C5DE6FA92D69C7ADD210A8E1029275D2ECB68353936916DCD9D3C228F415289CC801624BE06E494804438334DDB036AC9CA092A6

Something interesting I found is that there is [someone ](https://www.macos86.it/topic/3490-installer-yosemite-direct-download-link/)who apparently found the SHA-1 for this .pkg file as C542681961CC7FAFB9C15BFD4AF152A3A57F07FA, which is the same SHA-1 value I got for the .dmg file. But I found [another post](https://telemetr.io/en/channels/1010584557-AAAAADw8S-3Je31MI4h73Q/posts/1488) that cites the SHA-256 of the .dmg as DE869907CE4289FE948CBD2DEA7479FF9C369BBF47B06D5CB5290D78FB2932C6, which matches that of my .dmg file and is different from that of the .pkg file. Ergo, I'm confused if I should burn the .pkg file onto the USB drive, and if that's even possible given the instructions citing to use the .dmg file [here](https://github.com/balena-io/etcher/issues/1572#issuecomment-313395648). My attempts to create a bootable drive using createinstallmedia and Diskmaker X inside the VM have all ended in failure.

MrPhil17, 2020-12-13:
Etcher is not made for this purpuse. Etcher works only with Linux ISO. You need to use a program like TransMac for create a bootable usb drive from a Windows machine. The best way is to use a mac and createinstallmedia command on terminal.

Mennaruuk, 2020-12-13:
But [this comment](https://github.com/balena-io/etcher/issues/1572#issuecomment-313395648) has detailed instructions on how to create bootable Mac USB drives using Etcher. TransMac fails the task unfortunately, it does work with Mac ISOs, the last of which I think is Snow Leopard.