balena-io/etcher #3376: Error cloning 128GB SD card

bbiesele, 2020-12-06:
- **Etcher version:**  Version 1.5.111 (1.5.111)
- **Operating system and architecture:** Mac OS Catalina 10.15.7 (19H15)
- **Image flashed:** Copy NVIDIA Nano developer 
- **Do you see any meaningful error information in the DevTools?** 

{"stack":"RangeError [ERR_OUT_OF_RANGE]: The value of \"offset\" is out of range. It must be >= 0 && <= 1048576. Received 1048576\n    at read (internal/fs/promises.js:228:3)\n    at FileHandle.read (internal/fs/promises.js:88:12)\n    at BlockDevice.read (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:8178:48)\n    at BlockDevice.read (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:3413:30)\n    at SourceDisk._read (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:9844:38)\n    at SourceDisk.readAccordingToPlan (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:3012:32)\n    at async SourceDisk.read (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:2894:20)\n    at async ConfiguredSource.read (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:9900:20)\n    at async SparseReadStream.__read (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:3946:17)\n    at async SparseReadStream._read (/Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:3927:27)","message":"The value of \"offset\" is out of range. It must be >= 0 && <= 1048576. Received 1048576","name":"RangeError","code":"ERR_OUT_OF_RANGE","image":"disk9"}
console.<computed> @ gui.js:56894

Popup with error message 
[image]

shawaj, 2020-12-08:
@bbiesele could you update to v1.5.112 to see if that fixes the issue?

You can get it from [here](https://github.com/balena-io/etcher/releases/tag/v1.5.112) or just from the website [here](https://www.balena.io/etcher/)

bbiesele, 2020-12-09:
It worked with the same size card and the same image. But I don't know if
the state of the card was the same.

Thanks for the help. I'll let you know if it happens again; and see if I
can find what the difference is

shawaj, 2020-12-10:
@bbiesele ok great. will close this for now then. Feel free to reply back if any similar issues

closed 2020-12-10 by shawaj