balena-io/etcher #3378: Unable to flash to USB Sandisk Cruzer Glide 2.0 16GB Flash Drive

Erobles23, 2020-12-07:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

lurch, 2020-12-07:
>     * **Etcher version:**

???

>     * **Operating system and architecture:**

???

>     * **Image flashed:**

???

>     * **Do you see any meaningful error information in the DevTools?**

???

Erobles23, 2020-12-07:
I actually got passed all of that and I just shut down my computer, then held Command and chose "no security" and allowed booting from external media but after that I'm stuck. my computer just keeps going back to my usual account and starting up normally. What am I doing wrong?

shawaj, 2020-12-08:
@Erobles23 you haven't provided any of the requested information so it's very difficult to answer you.

Can you provide the requested information:
Etcher version:
Operating system and architecture:
Image flashed:
Do you see any meaningful error information in the DevTools?

shawaj, 2020-12-13:
@Erobles23 any update here?

lurch, 2020-12-14:
Perhaps @Erobles23 is just looking for https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable ? :wink:  (perhaps trying a different image-flashing tool will create a USB disk that his BIOS is actually able to boot?)

autumnadagio, 2020-12-15:
@Erobles23 I just flashed this exact USB drive with no issues, so I don't think it's the USB...

shawaj, 2020-12-15:
Will close this issue as no reply from @Erobles23

closed 2020-12-15 by shawaj