balena-io/etcher #3380: Addition of Etcher to PortableApps.com!

Peppernrino, 2020-12-08:
- **Etcher version:** 1.5.112+
- **Operating system and architecture:** Windows
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

I have packaged Etcher in .paf.exe format for PortableApps.com, and would like to finally inform you!

https://portableapps.com/node/59728

Let me know what you think. If there are no bugs, etc, we can put it on the list to be added to the official list of apps. :)

lurch, 2020-12-09:
"For some reason, the website version is halted at 1.5.45, but Github has newer versions."

Try enabling JavaScript on the Etcher website :wink: 

Is this different to the "portable" Etcher that Balena already provides?

Peppernrino, 2020-12-09:
javascript! lol oh damn! i use ublock origin and umatrix. sorry aboot that. will change that immediately. :P 

and no it is not. it is just packaged in PortableApps.com format. it has some NSIS things that move directories back and forth (for settings etc), clears up empty folders after itself... nothing about the original .exe is changed. it simply references it through some .ini files. i have turned off auto-update and verify in settings and saved that as DefaultData. :) 

the .ini files also keep the settings safe during an update.

Peppernrino, 2020-12-09:
ah. it was the <code>rawgithubusercontent.com</code> part. thanks again. :)

shawaj, 2020-12-16:
@Peppernrino thanks very much - is there testing that you need people to perform for this? If so, what do you need there?

I noticed it says it is in beta testing part of forum.

Is the use case for this just that it is a "neater" way of running portable apps and having them all located in a single place?

Peppernrino, 2020-12-27:
I mostly just need your blessing... if there are any anti-.exe-hacking thingies you want to run it through... make sure I didn't add my company's name to the credits (looking at you, dude that packaged Arduino.... look forward to a repack from me today lol...) etc.

It is in the beta testing forum, because that is the release process of the apps on this platform. You get/make a request for an app to be portablized, fill the request with a reference to the request in the help.html(? unsure rn. don't feel like checking) and then submit it to beta testing for people to find problems with it... trailing files or empty folders, registry entries mishandled, stuff like that.

And yes, but not really. It also provides users a portal to update the app without having to visit a download page, for one. I've been using portableapps.com for years. I just started contributing because I like it so much lol. There are other portableapps platforms, but this one is fairly well-managed and the UI is cool. Like a little Start menu. :)

shawaj, 2021-01-02:
@Peppernrino makes sense! I don't see any issue with it being on there personally, but I guess it's not really up to me. I'm not a lawyer, but the [license](https://github.com/balena-io/etcher/blob/master/LICENSE) seems to grant you the ability to distribute it etc so don't think you need to ask for permission per se. But it's certainly appreciated.

I guess there is unlikely to be any appetite to have this as an "official" source because of the extra work of updating another platform with the installers etc. however if you are happy to maintain it and it helps people to access Etcher then I can't see any issue with it really.

Perhaps we can leave this issue open for now, and once the app is officially accepted after the beta we can close the issue.