balena-io/etcher #3387: I can't Flash the "jetson-nano-4gb-jp441-sd-card-image" to the drive

Flyooofly, 2020-12-20:
- **Etcher version:** the most new version
- **Operating system and architecture:** windows
- **Image flashed:** jetson-nano-4gb-jp441-sd-card-image
- **Do you see any meaningful error information in the DevTools?** No
I have checked the interfate is usb3.0 and I don't know why I can't success

lurch, 2020-12-20:
> Etcher version: the most new version 1.5.45

You're only "seeing" that as the newest version because you've got javascript turned off! Either turn on JavaScript for the Etcher website, or download the latest version from https://github.com/balena-io/etcher/releases

Flyooofly, 2020-12-20:
> > Etcher version: the most new version 1.5.45
> 
> You're only "seeing" that as the newest version because you've got javascript turned off! Either turn on JavaScript for the Etcher website, or download the latest version from https://github.com/balena-io/etcher/releases

Oh my god..I'm stupid..It works ,thank you!

closed 2020-12-20 by Flyooofly

lurch, 2020-12-20:
@thundron @zvin @shawaj There have been several times this has happened now... perhaps it's either worth updating the Etcher version displayed on the no-javascript version of the Etcher website to a more recent version; or maybe add some wording to the no-javascript version of the website to say that it's probably an old version, and if users are having problems they should see if there's a later version available at https://github.com/balena-io/etcher/releases ? (or maybe just add some text telling users that they should turn on javascript to see the latest Etcher version??)

According to https://github.com/balena-io/etcher/blob/master/CHANGELOG.md Etcher v1.5.45 was released over 18 months ago!!!

shawaj, 2020-12-20:
@lurch good idea - only issue is I don't know how the etcher.io page is updated with a default version.

@zvin @thundron any idea about this?

shawaj, 2020-12-20:
See #3380

thundron, 2020-12-22:
we have a PR that updates the fallback version to one more recent, but maybe an additional message is better so that we also instruct the user that it's an old version regardless

shawaj, 2020-12-22:
@thundron is that page generated in a different repo?

Definitely think that would be good about mentioning it may be an old version.