balena-io/etcher #3388: I can not return the flash drive to its original state

mrkaban, 2020-12-20:
- Etcher version: balenaEtcher-Portable-1.5.113 (tried and not portable version)
- Operating system and architecture: Windows 10 x64
- Image flashed: rescuezilla-2.1-64bit.groovy, Linux mint
[image]
After using Etcher to write a disk image to a USB Flash drive, I am faced with the problem of restoring the original disk state. This is not related to the disk image, this is not the first time this problem and the distributions were different. By means of Windows OS it is possible to format only one of the three volumes created on the basis of one flash drive. For full recovery, you have to shaman almost using the Acronis software. Make a button in the program to restore the normal state of the flash drive.

shawaj, 2020-12-20:
@mrkaban what exactly do you mean by restoring "the original disk state"?

lurch, 2020-12-21:
@shawaj He probably means a single NTFS or exFAT partition, spanning the entire disk?

This might be a duplicate of #273

mrkaban, 2020-12-21:
The point is that before using the program there was 1 flash drive = 1 section, and after using the program 1 flash drive = 3 sections, and two of them I cannot delete, in order to make one single section again. Look, I right-click, but there is no delete item.
[image]

lurch, 2020-12-21:
Does https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives work?

mrkaban, 2020-12-22:
Yes, it helped, thanks! Maybe you can implement this inside the program? it's just hard to look for instructions every time

closed 2020-12-22 by mrkaban

shawaj, 2021-01-02:
@mrkaban glad you got it working. Often the best thing to do is to download the official SD card formatter tool https://www.sdcard.org/downloads/formatter/

And reformat the drive.

The difficulty with having this stuff inside the Etcher application is that there are so many different scenarios. But thanks for the idea in any case.

maxmod, 2021-05-19:
The FormatUsb tool is a reliable way to format USB flash drives, such as USB keys/pendrives, memory sticks,SD Memory Card, SDHC Memory Card and SDXC Memory Card SD/SDHC/SDXC Cards.
https://formatusb.com/download

issu-shooter, 2021-05-27:
Fedora's flashing tool has such a function, not sure how many scenarios it can cover, but simply reverting a successful installation of an ISO from a common linux distribution should be easily done.