balena-io/etcher #3390: Etcher not work on macOS Catalina

zhuyingda, 2020-12-21:
- **Etcher version:** 1.5.45
- **Operating system and architecture:** MacOS Catalina 10.15.7
- **Image flashed:** gladys-4.0.0.img (download and unzip from https://github.com/gladysassistant/gladys)
- **Do you see any meaningful error information in the DevTools?** 

`Mon Dec 21 2020 08:41:39 GMT+0800 (CST) Flash error ({"error":{"name":"Error","message":"EPERM: operation not permitted, open '/dev/rdisk2'","stack":"Error: EPERM: operation not permitted, open '/dev/rdisk2'","code":"EPERM","syscall":"open","errno":-1},"applicationSessionUuid":"9a3b459e-95b4-4c0f-bd4b-cae7e941ee1c","flashingWorkflowUuid":"1ea72e03-e61a-490e-b28c-c40ef0ae0d9f","flashInstanceUuid":"73918c68-761c-48e9-8c6a-1e2fe86c6a76","image":"/Applications/balenaEtcher.app/gladys-4.0.0.img","drives":[{"size":63864569856,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/PEG1@1,1/UPSB@0/DSB2@2/XHC2@0","mountpoints":[],"description":"SanDisk SDDR-B531 Media","isUAS":null}],"driveCount":1,"uuid":"73918c68-761c-48e9-8c6a-1e2fe86c6a76","unmountOnSuccess":true,"validateWriteOnSuccess":true,"sample":0.1})`

I have tried the way sudo execute the etcher directly which I see here: [2833-511152047](https://github.com/balena-io/etcher/issues/2833#issuecomment-511152047)

It solved the problem that I can not open the img file because of permission, but it does not solved the problem that I can not open the `/dev/rdisk2` usb device (it is a sdcard reader/writer).

I also have tried [2833-534984500](https://github.com/balena-io/etcher/issues/2833#issuecomment-534984500) open "Full Disk Access" for etcher.

It still does not solved my problem.

Anyone can help me?

shawaj, 2020-12-21:
Please try v1.5.113 as mentioned in the other thread and then hopefully this should work as well https://github.com/balena-io/etcher/releases/tag/v1.5.113

zhuyingda, 2020-12-21:
Solved. 
https://github.com/balena-io/etcher/issues/2833#issuecomment-748726970

Thanks, @shawaj

shawaj, 2020-12-21:
My pleasure 😁

Let us know if we can help with anything else.

closed 2020-12-21 by shawaj