balena-io/etcher #3395: Unable to scroll drive list

mauricev, 2020-12-25:
- **Etcher version:** 
- 1.5.113
- **Operating system and architecture:** 
- 10.14.6
- **Image flashed:** 
- affects all images

This bug has been reported twice before 
https://github.com/balena-io/etcher/issues/789
https://github.com/balena-io/etcher/issues/202

So it apparently had been fixed at one point, but now it's back.

See this picture:

[image]

It's actually three bugs. There is no scroll bar. Up and down arrows don't work and it shows drives that we never use, even labeling them "large drives".

thundron, 2020-12-29:
You can use the mouse wheel or a touch screen to scroll the list

thundron, 2020-12-29:
As for the large drives, we can't possibly know if you use them or not. Etcher shows all available drives

closed 2020-12-29 by thundron

lurch, 2020-12-29:
"You can use the mouse wheel or a touch screen to scroll the list" - I guess that doesn't help people who have neither a touch screen nor a mouse wheel? :thinking:

thundron, 2020-12-31:
Is it actually possible to have a mouse without a wheel? 🤔  I just assumed it's a standard feature nowaday, isn't it?

mauricev, 2020-12-31:
Ah, it’s a broken interface. I would have never guessed to use the scroll on the mouse.