balena-io/etcher #3396: Trying to Select Target (Usb) For Tails Image

kaleb986769, 2020-12-30:
(Etcher) Hello, I am currently 5 seconds away from having a stress induced seizure from racking my brain trying to select my usb after finally uploading a tails image to Etcher that has a partition table. When I click Target my usb pops up but is not clickable, only my main drive is under hidden. Please help for I am technologically inept.
[image]

thundron, 2020-12-31:
The `source drive` label is for whenever the target you're trying to access also contains the source that you want to flash, thus making it impossible to flash it.
Try moving the source outside of the target and re-selecting it from outside. That should make the target available for flash

sajolida, 2021-12-22:
I think that this issue should be closed as not actionable and the root problem of bad feedback addressed on #3662.