balena-io/etcher #3397: I can't launch etcher

seigiakatsuka, 2021-01-02:
Not sure what this is asking I've googled it and tried to correct the issue but I'm at a loss
- **Etcher version 1.5.113:** 
- **kali linux 20.3** 
- **$ ./balenaEtcher-1.5.113-x64.AppImage 
[23013:0101/215433.131410:FATAL:setuid_sandbox_host.cc(158)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that /tmp/.mount_balenaiHJ1pK/chrome-sandbox is owned by root and has mode 4755.
/tmp/.mount_balenaiHJ1pK/balena-etcher-electron: line 10: 23013 Trace/breakpoint trap   "${script_dir}"/balena-etcher-electron.bin "$@"
          **

lurch, 2021-01-02:
Duplicate of #3293 ?

seigiakatsuka, 2021-01-06:
> Duplicate of #3293 ?

so I wasn't sure what I was looking at, are those solutions or just other people who had the issue?

EdoardoLopez, 2021-01-13:
OS: Debian GNU/Linux 10 (buster)

running with `sudo` into terminal show message `No protocol specified`

thundron, 2021-01-15:
Moved the discussion to https://github.com/balena-io/etcher/issues/3129. Closing

closed 2021-01-15 by thundron