balena-io/etcher #3398: Device to small to write image of same device

Muller59, 2021-01-03:
- **Etcher version:1.5.80**
- **Operating system and architecture:win10 x64**
- **Image flashed: unix sd card image created by dd**
- **Do you see any meaningful error information in the DevTools? no**

i create a backup image of a raspberry pi 32 Gb SD card with the command
dd if=/dev/mmcblk0 of=//NAS/backup.img bs=1M conv=sync,noerror iflag=fullblock
the result file is 29,7 GB (31.915.507.712 bytes) 

when i try to use etcher to write this file back to the SAME SD card, etcher tells me the SD card is to small.

also tried with the latest version 1.5.113, same result.
how to resolve this?

lurch, 2021-01-03:
I did a quick `dd --help` and I see: "fullblock  accumulate full blocks of input (iflag only)"

So my _guess_ is that this causes `dd` to pad the output file to be an exact multiple of 512 bytes (e.g. by appending extra zeros), but your SD card _isn't_ an exact multiple of 512 bytes, which is why Etcher (correctly) complains that your backup image is now bigger than your SD card.
If my guess is correct, the workaround should be to create your backup without using the `iflag=fullblock` option. Or I guess you could use the `truncate` command to shrink your existing backup file to the exact size of your SD card?

Muller59, 2021-01-04:
Andrew,

 

Thanks for the reply and suggestion

I will test today without the fullblock option.

 

 

Regards,

 

Rob Muller

mailto:r.b.muller@gmail.com

lurch, 2021-01-04:
Just a quick reply to let you know that you've publicly posted your email address to https://github.com/balena-io/etcher/issues/3398