balena-io/etcher #3399: Keep getting flash failed when trying to a flash file on to a sd card

d0llarbills, 2021-01-04:
I am trying to flash a file on to my sd card but keep getting the flash failed message. can someone help?

I have bought a new sd card reader and sd card but that has not worked.
[image]

appreciate all help in advance thanks :)

thundron, 2021-01-04:
Please fill in the information in the template, there are some steps to help you gather more information on the issue to provide some context

mrocach, 2021-01-16:
Hi. 
I have the same problem. The first time I could flash a sd card. After the third time, and the next times , didn't work. Even with the one sd card I had worked it before. I tried with SanDiskUltra HCI 10 (32GB y 64GB) and Kingston SDC 10 (32 y 64GB).

I would appreciate any help you could provide. Thanks.

nocom, 2021-01-18:
I'm getting the same problem, running macOS 11.1 (Big Sur) and Etcher telling me each time that the file is eventually corrupted.

sergiomoraes, 2021-01-23:
I suggest you try running the Etcher as Administrator. I was having the same issue but that solved my problem.

On previous versions, Etcher would throw a warning *before* starting the flashing process saying that it couldn't proceed without Admin permissions, but that is not happening on the latest version for some reason.

nocom, 2021-01-23:
Im sorry, I thought I had posted that everything was ok after updating Etcher to the actual version. GreetingsAm 23.01.21 um 19:34 schrieb Sergio Moraes
                
            
            
                
                    Von: "Sergio Moraes" <notifications@github.com>Datum: 23. Januar 2021An: "balena-io/etcher" <etcher@noreply.github.com>Cc: "nocom" <ebussmann@web.de>,"Comment" <comment@noreply.github.com>Betreff: Re: [balena-io/etcher] Keep getting flash failed when trying to a flash file on to a sd card (#3399)
                
                
I suggest you try running the Etcher as Administrator. I was having the same issue but that solved my problem.
On previous versions, Etcher would throw a warning before starting the flashing process saying that it couldn't proceed without Admin permissions, but that is not happening on the latest version for some reason.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/balena-io/etcher/issues/3399#issuecomment-766157709",
"url": "https://github.com/balena-io/etcher/issues/3399#issuecomment-766157709",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

Ahmad1z1, 2021-02-17:
I keep getting the exact error, using version 1.5.116,  It doesn't pass the validation step.  the drive seems to work fine on windows but my 905x, 2gb box does not recognize it.  Keeps telling me its corrupted.  I’ve tried to run Etcher in admin but that still keeps giving Flash failed.  

[image]

al1998petridis, 2021-03-28:
Same problem here, it goes to 100% of Validating and then fails. But if I skip validating, no warnings or errors are displayed.

fearixarmada, 2021-05-17:
Hi, I had this problem and it was being caused because controlled folder access was switched on in Microsoft Defender and I wasn't getting the notifications because Focus Assist was also on - Note. I download to my Desktop and had added this to my controlled access but I'm sure the download folder is on by default, hope that helps

LiNuBeLM, 2021-06-18:
Still the same problem

thundron, 2021-07-07:
does @fearixarmada 's solution help by chance?

nhphuong91, 2021-07-08:
[image]
Press `Ctrl+Shift+I` and look for log in `Console` tab. Mine shows that it's just an `Error: Unmount failed` so I can safely ignore it. It's just quite annoying to see something FAIL :))

thundron, 2021-07-13:
true @nhphuong91 , in fact it's interesting that there's no error modal accessible as it should provide the error message you see in the console, so something isn't right for sure there

lurch, 2022-02-02:
> I use a USB port hub on my monitor (Asus PA279CV). It has 2 modes: USB2.0 & USB3.1. If I flash my SD card in USB2.0 mode, it's success. However, If I switch to USB3.1 mode, the flashing process end with Error: Unmount failed log.

Curious, USB is supposed to be forwards- and backwards-compatible in both directions. Perhaps your SD reader is actually USB3-capable, and uses [UASP](https://en.wikipedia.org/wiki/USB_Attached_SCSI) when it's connected via USB3, but not when via USB2? (and that's causing the USB storage drivers in Windows to behave differently?)

Andrewn2, 2022-02-25:
> Hi, I had this problem and it was being caused because controlled folder access was switched on in Microsoft Defender and I wasn't getting the notifications because Focus Assist was also on - Note. [...]

THIS WAS MY SOLUTION!!

hbsbh81, 2022-04-02:
> I encountered this problem after I enabled Windows' Ransomware protection. The way I got Balena to work was to let it fail, then open Windows Protection History (Win+R `explorer windowsdefender://History`) and choose action "Allow on device" for the most recent item, a "Protected memory access blocked" instance. Try etching again. Hope it helps others.

THIS!

azhikin, 2022-04-20:
I had this problem trying to flash used cards for PiAware when trying to update to 7.2. I had to remove the partitions on the SD card (Win 10 or 11, search "create and format hard disk partitions", right-click on each volume and select "delete" until there's only one volume), then balena-Etcher worked just fine.

kelvingarger, 2022-10-21:
> I suggest you try running the Etcher as Administrator. I was having the same issue but that solved my problem.
>
> On previous versions, Etcher would throw a warning _before_ starting the flashing [...]

thanks a lot my friend... it worked smoothly after running belena with admin permissions. cheers

WaffyCactus, 2022-12-06:
Try turning off the Windows Defender it worked for also don't forget to turn on the Windows Defender back after the purpose is full filled 

Cheers 🤗

chadericstevens, 2023-01-07:
I had the same failure problem.  Tried several things.  I took my sd card and card reader and plugged it directly into my comuter's USB port.  Previously it was plugged into a USB 3.0 4 port hub.  Now it is writing the image correctly.

stonegrimm, 2023-02-06:
Keep getting this error too, nothing on Windows Defender history related to the application. On Windows 11

casivaagustin, 2023-02-09:
On windows 11 you have to disable controlled folder access in order to make it work.

https://forums.balena.io/t/unable-to-use-etcher-with-win11/349919/7

pilotwithnoITskills, 2023-05-19:
disabled controlled folder was already switched off, i switched off windows defender firewall, used etcher as an administrator to flash the image on my SD card, used 3 different SD cards, and still get the message during verification stage: flash failed....
Any suggestions?
If i put the SDcard with the failed flash, in the pi3b, the stratux worked during a few weeks, now i get after a while error message gps disconnected not communicating. Changed GPS mouse, same issue. Also the wifi tends to stop working, then connceting again... I use iphone and ipad to check, they both cant find the wifi from stratux anymore. After reboot of stratux, it works again for a short period, and then the issues start again....
What could be wrong? In the beginning it all worked ok...

bcutter, 2024-01-07:
Also fails. Win 11, tested with several target drives, same outcome. Write is actually successful (target drive written fine) but the error rises up anyway everytime.

chrisdafer, 2024-03-29:
TLDR, do yourself a favor and try using a USB 2.0 port directly on your machine.

Windows 10 here, and I was having this issue and tried numerous devices including flash drive, microSD card, card reader, USB hub, USB port directly on the machine, I did not have controlled folder on and I was running with Admin rights. Flash drive failed on validation, but microSD card failed instantly. I was just now able to successfully flash both by......plugging the flash drive and the SD card reader directly into a USB 2.0 port on the computer. There must be a compatibility issue with Etcher. This was a fresh installation of Etcher, too.

Polpix, 2024-05-15:
I had the same pb with the USB key plugged on a hub-power supply.
It re-failed in Admin mode
It succeeded in Admin mode, plugged directly in the pc.

vinser, 2024-09-23:
> I keep getting the exact error, using version 1.5.116, It doesn't pass the validation step. the drive seems to work fine on windows but my 905x, 2gb box does not recognize it. Keeps telling me its corrupted. I’ve tried to run Etcher in admin but that still keeps giving Flash failed.
> 
> [image]

You should set the windows defender real-time protection exception for etcher and it will go right. See https://support.microsoft.com/en-us/topic/how-to-add-a-file-type-or-process-exclusion-to-windows-security-e524cbc2-3975-63c2-f9d1-7c2eb5331e53

FernandoDDav, 2025-02-10:
Hello,
I'm not sure if my input addresses/resolves the presented issue (did not take the time to read the whole list of responses) and still...
I spent today 7 hours trying to create a bootable SD using Etcher without success till, in one post I read, there was a unintended hint to the problem (at least in my case).
For many years my computers had either two physical disks or two partitions (C and D), C for all related to system and standard software while D for all my data and programs I developed.
My current laptop is equipped with a 1TB SSD and hence there is no room/point in creating partitions. Instead, I create a "link" device (I named it DISK_D and assigned to "D:") which points to a folder within C.
Now, when trying to flash the SD I ran the program from D:...\My_Apps and picked the image file from D:...\Disk_Images. I tried this scenario tens of times enabling and disabling the anti-virus, running as Admin, etc. without success.
The abovementioned hint made my try one more time, but now running from C:\Disk_D\My_Apps for the program and C:\Disk_D\Disk_Images and this time it worked like a charm.

I hope this turns useful for someone.

lurch, 2025-02-13:
@FernandoDDav I believe that'll be because your "link drive" is only visible to the current user, and not to the "elevated user" that Etcher needs to use to do low-level block writes to your SD card. (I.e. the elevated user can only see `C:` but not `D:`)

FernandoDDav, 2025-02-13:
Thank you @Andrewn2 for your comments.
I just tested your claim (elevated user would not see D:) and you are absolutely right!!!!
Though I'm working with Windows for very many years, faced this issue in the past a number of times resolving them by using C: as root, I never took the time to investigate why with D: it doesn't work.
Now, thanks to you, I know.
Cheers Sir!!