balena-io/etcher #3400: built from source wrong at last when building on arm64 arch system.

daozhu, 2021-01-05:
- **Etcher version: 1.5.113 (bulit from source) : 2021.01.05 14:00:00
- **Operating system and architecture: debian , arm64
- **Image flashed: not yet
- **Do you see any meaningful error information in the DevTools?** 

the error info were: 
```
• writing effective config  file=dist/builder-effective-config.yaml
  • packaging       platform=linux arch=arm64 electron=9.4.0 appOutDir=dist/linux-arm64-unpacked
  • downloading     url=https://github.com/electron/electron/releases/download/v9.4.0/electron-v9.4.0-linux-arm64.zip size=77 MB parts=8
  • downloaded      url=https://github.com/electron/electron/releases/download/v9.4.0/electron-v9.4.0-linux-arm64.zip duration=2m47.493s
  • asar usage is disabled — this is strongly not recommended  solution=enable asar and use asarUnpack to unpack files that must be externally available
  • building        target=deb arch=arm64 file=dist/balena-etcher_1.5.113_arm64.deb
  • downloading     url=https://github.com/electron-userland/electron-builder-binaries/releases/download/fpm-1.9.3-2.3.1-linux-x86/fpm-1.9.3-2.3.1-linux-x86.7z size=4.6 MB parts=1
  • downloaded      url=https://github.com/electron-userland/electron-builder-binaries/releases/download/fpm-1.9.3-2.3.1-linux-x86/fpm-1.9.3-2.3.1-linux-x86.7z duration=1m6.266s
  ⨯ cannot execute  cause=exit status 1
```

my sys arch is arm64 , but it download the x86 arch binaries. 
> - fpm-1.9.3-2.3.1-linux-x86

lurch, 2021-01-05:
Probably related to https://github.com/electron-userland/electron-builder-binaries/issues/17 ?

daozhu, 2021-01-06:
but i can build success on my mac:
```
edz@daozhudeMacBook-Pro-8 etcher % PATH=$(pwd)/node_modules/.bin:$PATH electron-builder --linux deb --arm64
  • electron-builder  version=22.9.1 os=19.6.0
  • loaded configuration  file=/Users/edz/work/sys-proes/etcher/electron-builder.yml
  • electron-rebuild not required if you use electron-builder, please consider to remove excess dependency from devDependencies

To ensure your native dependencies are always matched electron version, simply add script `"postinstall": "electron-builder install-app-deps" to your `package.json`
  • writing effective config  file=dist/builder-effective-config.yaml
  • packaging       platform=linux arch=arm64 electron=9.4.0 appOutDir=dist/linux-arm64-unpacked
  • downloading     url=https://github.com/electron/electron/releases/download/v9.4.0/electron-v9.4.0-linux-arm64.zip size=77 MB parts=8
  • downloaded      url=https://github.com/electron/electron/releases/download/v9.4.0/electron-v9.4.0-linux-arm64.zip duration=4m10.145s
  • asar usage is disabled — this is strongly not recommended  solution=enable asar and use asarUnpack to unpack files that must be externally available
  • building        target=deb arch=arm64 file=dist/balena-etcher_1.5.113_arm64.deb
  • downloading     url=https://github.com/electron-userland/electron-builder-binaries/releases/download/linux-tools-mac-10.12.3/linux-tools-mac-10.12.3.7z size=520 kB parts=1
  • downloaded      url=https://github.com/electron-userland/electron-builder-binaries/releases/download/linux-tools-mac-10.12.3/linux-tools-mac-10.12.3.7z duration=17.035s
  • downloading     url=https://github.com/electron-userland/electron-builder-binaries/releases/download/fpm-1.9.3-20150715-2.2.2-mac/fpm-1.9.3-20150715-2.2.2-mac.7z size=5.4 MB parts=1
  • downloaded      url=https://github.com/electron-userland/electron-builder-binaries/releases/download/fpm-1.9.3-20150715-2.2.2-mac/fpm-1.9.3-20150715-2.2.2-mac.7z duration=13.974s
edz@daozhudeMacBook-Pro-8 etcher % 
edz@daozhudeMacBook-Pro-8 etcher % 
edz@daozhudeMacBook-Pro-8 etcher % ls
CHANGELOG.md		Makefile		afterSignHook.js	dictionary.txt		entitlements.mac.plist	npm-shrinkwrap.json	scripts			typings
CODEOWNERS		README.md		assets			dist			generated		package.json		tests			webpack.config.ts
FAQ.md			SUPPORT.md		beforeBuild.js		docs			lib			repo.yml		tsconfig.json
LICENSE			afterPack.js		dev-app-update.yml	electron-builder.yml	node_modules		requirements.txt	tsconfig.webpack.json
edz@daozhudeMacBook-Pro-8 etcher % cd dist 
edz@daozhudeMacBook-Pro-8 dist % ls
balena-etcher_1.5.113_arm64.deb	builder-effective-config.yaml	linux-arm64-unpacked
```

marceloch2, 2021-09-15:
https://github.com/jordansissel/fpm/issues/1801

PR #4132 (2023-12-21): Modernize build pipeline
Switch to electron-forge which is now officially adopted by the Electron project. Solves a bunch of issues along the way.

Issues related to experimental v1.18.13:

Closes #4138 
Closes #4129
Closes #4119 

Build related issues:

Closes #4136 
Closes #4125 
Closes #4103 
Closes #4087 
Closes #3767 
Closes #3728 
Closes #3653 
Closes #3400 
Closes #3292 
Closes #3269 
Closes #3263 
Closes #2870 
Closes #2826 

(Admittedly, some of the issues are linked opportunistically -- please reopen them if the issue is not actually solved)
closed 2023-12-21 by dfunckt