balena-io/etcher #3404: Etcher replaced F1 Help function

Afrothetics, 2021-01-11:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

Afrothetics, 2021-01-11:
Hi, this is an issue that I can't replicate, or, more to the point, haven't tried to replicate, but I thought the community should know. For several days, I tried to call up help from inside an application using the F1 key, but Etcher would appear. So, today I decided to correct this.

FYI, I'm using Linux Mint 20.04 OS. First, I launched Settings, selected keyboard, selected the "Layout" tab, selected my keyboard from the left-hand panel, then clicked "Reset to Defaults". Rebooted, opened an application (LibreOffice Writer), and hit F1. Same problem. Then, I opened Synaptic and did a complete uninstall of Balena-Etcher_1.5.109. That solved the issue. 

I'm now using an AppImage version of Balena-Etcher and there is no conflict...yet. (:-)

lurch, 2021-01-11:
How weird, I wonder if it might be related somehow to #3374 ?

diyhouse, 2021-09-24:
I have exactly the same problem,.. calling up help within LibreOffice Help,.. etcher is started,.. following a screen which states  a warning screen that 'Missing partition table',.. it look like this is not a bootable image,..etc etc
Weird!!

thundron, 2021-09-24:
It is highly likely a duplicate of the other issue, reason being that probably it tries to open a URL, and instead Etcher is opened.
We will check if anything can be done on our part, but for now the best is to monitor the xdg-utils issue that has been linked in the other discussion. Closing

closed 2021-09-24 by thundron