balena-io/etcher #3415: [Feature Request] flash to only a partition not the whole drive

scratchusernamemrtbts, 2021-01-20:
- **Etcher version:** 
- 1.5.115
- **Operating system and architecture:** 
- Mac OS 11.2 beta 2 intel core i5 iMac 2017 27"
- **Image flashed:** 
- raspberry pi dekstop
- **Do you see any meaningful error information in the DevTools?** 
- no

make we can select to flash to only a partition of a drive
I have a 2 TB drive and I want to flash the raspberry pi desktop to it but I already have Time Machine backups on it so I don't want to erase it just want to flash to a partition that contains nothing

lurch, 2021-01-20:
Raspberry Pi Desktop consists of two partitions (boot and root) and thus *needs* to be written to the whole drive (i.e. overwriting the MBR).

But see also #3118

lurch, 2021-01-20:
That sounds like the job of a fully-fledged partition manager tool https://www.google.com/search?q=partition+manager+mac rather than a simple disk-image-flasher like Etcher. Entirely IMHO of course.

thundron, 2021-02-02:
it's already been partially discussed in https://github.com/balena-io/etcher/issues/414#issuecomment-219723312 - it's outside of the scope of Etcher, but we were considering the addition of a plugin system in the future

closed 2021-02-02 by thundron