balena-io/etcher #3417: Feature Request - To change the drive name

maifeeulasad, 2021-01-21:
When flashing a drive, be able to give the drive some name. That would be really fun.

lurch, 2021-01-22:
But I guess there's no way of knowing if the OS image being written by Etcher might rely on a specific drive name; so it's (theoretically) possible that if Etcher changes the name of the drive, that might cause the flashed OS to fail to boot? :shrug:

thundron, 2021-01-22:
@lurch is it actually possible for an image to rely on a drive's name though? 🤷 it doesn't sound fail-safe at all, so it's strange to me that such thing is possible

lurch, 2021-01-22:
I dunno, that's why I said "(theoretically) possible". :wink:  I know that e.g. Raspberry Pi OS identifies its partitions by UUID (and references them as such in its `/etc/fstab`), so I was wondering if perhaps some other OSes might identify partitions by label? :shrug: